rstrouse/ESPSomfy-RTS #210: [Request] Add current shade position to MQTT birth message

Calimerorulez, 2023-12-24:
Hi,

Is it possible to add the current shade position/percentage to the MQTT birth message (I believe it is the birth message)?

Clients connecting to the broker only see the IP, host, firmware version, serverID and status when connecting. Only when I move the shades the position is reported.

rstrouse, 2023-12-24:
If you use the Github update and select main for the version it will now retain the state information.   It should give you the last message when you connect to the broker.  The AD messages however are not retained.  If that is required it will take some tweaking to kill these but I suspect the MQTT AD client will probably choke on that if I need to clear AD messages.

Calimerorulez, 2023-12-27:
Works fine. Thanks!

closed 2023-12-27 by Calimerorulez