rstrouse/ESPSomfy-RTS #211: [Request] Add ESP details in  HA integration

GelsemaP, 2023-12-24:
Hi, I would like to request some more info in the integration in HA. Mainly driven because I am lazy but also that i have so many IOT devices and cannot remember IP addresses of each.

I wanted to view the details of the ESPboard but I couldnt remember the IP address assigned to it. My IOT devices are on a different network in which the HA instance has an interface in. From my laptop on a more trusted network I allowed management ports to IOT devices. I went into the HA integration and wanted to see the IP, or a link to the ESPboard. I couldnt find it and clicking on configure, entities or devices just caused circles....
I would love to have information on the ESP device in the HA integration, with at least IP address, Wifi strength and SSID as a nice bonus.
Am willing to buy a coffee ;-)

thanks! and happy holidays.

rstrouse, 2023-12-24:
I'll have a look.  I have no issue putting information into the device info.  However, the problem is getting it to display somewhere in the HA interface.  However, if you want to see the IP address for the device you can simply click on the configure button and it will appear.

[image]

GelsemaP, 2023-12-24:
and now I feel dumb. I thought that I had tried all buttons, options and/configurations. And what I remembered was that i was getting in circles, from devices to overview to entities to device... Thanks for pointing it out. I must say that for my request I am using ESPhome based devices as example.

rstrouse, 2023-12-24:
Yeah the hub data isn't organized very well in HA.  But I will add some diagnostic categories in the next release that will show up as entities for the configuration.  This will likely include the SSID and the Wifi RSSI.

rstrouse, 2024-01-01:
If you update to v2.3.0 for both the firmware and HA integration you will now have some cool diagnostic entities that will tell you everything you need to know about your ESPSomfy RTS device.  This includes the current Wifi strength, IP address, wireless SSID/channel, and the ESP32 chip type.

GelsemaP, 2024-01-03:
Works like a champ! thanks for adding this! Thanks a lot.

closed 2024-01-03 by GelsemaP