rstrouse/ESPSomfy-RTS #212: Only "Prog" button does not work...

Jezza34000, 2023-12-25:
Hello, 
I got something strange.
I can control my shutters with all control UP, DOWN & MY, everything work perfectly except "Prog" button.
I've try long press, short press, my shutters doesn't jog.

I have a RTL-SDR and i can see the Prog command sended :

```
{"time" : "2023-12-25 16:07:01.020814", "model" : "Somfy-RTS", "id" : 1146889, "control" : "Prog (8)", "counter" : 625, "retransmission" : 0, "mic" : "CHECKSUM", "mod" : "ASK", "freq" : 433.492, "rssi" : -0.132, "snr" : 15.375, "noise" : -15.507}
{"time" : "2023-12-25 16:07:01.020814", "model" : "Somfy-RTS", "id" : 1146889, "control" : "Prog (8)", "counter" : 625, "retransmission" : 1, "mic" : "CHECKSUM", "mod" : "ASK", "freq" : 433.492, "rssi" : -0.132, "snr" : 15.375, "noise" : -15.507}
{"time" : "2023-12-25 16:07:01.020814", "model" : "Somfy-RTS", "id" : 1146889, "control" : "Prog (8)", "counter" : 625, "retransmission" : 1, "mic" : "CHECKSUM", "mod" : "ASK", "freq" : 433.492, "rssi" : -0.132, "snr" : 15.375, "noise" : -15.507}
{"time" : "2023-12-25 16:07:02.993659", "model" : "Somfy-RTS", "id" : 1146889, "control" : "Prog (8)", "counter" : 626, "retransmission" : 0, "mic" : "CHECKSUM", "mod" : "ASK", "freq" : 433.493, "rssi" : -0.095, "snr" : 15.255, "noise" : -15.350}
{"time" : "2023-12-25 16:07:02.993659", "model" : "Somfy-RTS", "id" : 1146889, "control" : "Prog (8)", "counter" : 626, "retransmission" : 1, "mic" : "CHECKSUM", "mod" : "ASK", "freq" : 433.493, "rssi" : -0.095, "snr" : 15.255, "noise" : -15.350}

but no shutters reaction. After i've try with remote and shutter jog on Prog's button press. I'm a bit confused

Do you have an idea ?

rstrouse, 2023-12-25:
The up, down, and my buttons should not work until you have paired ESPSomfy RTS with the motor using the prog button.

Please make sure the addresses used in ESPSomfy RTS are unique and do not match any other remote that you are using.  ESPSomfy RTS is either using an existing remote address which is how you can use the up, down, and my buttons or it was successfully paired ESPSomfy RTS to the motor at one time with the Prog button.

Jezza34000, 2023-12-25:
My shutters are already paired and work. I notice this "Prog" problem when i want to create a group.
But for information, i did not "pair" be conventionnal method. I was using previsouly RFPLAYER and i migrate my remote address code & rolling into ESPSomfy RTS.
I'm using it now since 2 month and everything work perfectly, except Prog command... Really weird!

rstrouse, 2023-12-25:
Do you have Somfy branded motors or are they some other manufacturer?

Jezza34000, 2023-12-25:
It's Simu

rstrouse, 2023-12-25:
Look in your manual.  It is possible that there is some other button combination to put it into programming mode.  I saw one where you hold down the UP+DOWN buttons to get it to react.  If there is some other combination you can use the Virtual Remote and get the same process completed.

ViViDboarder, 2023-12-27:
I have the same issue with mine. The Prog button on my remote works fine, but not the virtual one.

I figure I can probably manually create groups by using the remote to trigger the program mode, but it could help others to figure it out. 

Anything I can do on my end to help debug?

Jezza34000, 2023-12-27:
OK @ViViDboarder let's investigate this subject... I have several questions:
- What brand/type of shutter do you have?
- If your "Prog" button does not work, how did you pair your shutters?
- What protocol do you use? RTS RTV RTW... 56/80bits?

Jezza34000, 2023-12-27:
OK after many and many test, i'm more and more confused : 

1. Test with PROG button from ESPSomfy : 
(Frame is received from an another ESP32)
```
[{"encKey": 162,
  "address": 15236113,
  "rcode": 642,
  "command": "Prog",
  "rssi": -28,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [245929,122764,25430,10285,7325,2615,2516,2618,2508,2595,2546,2583,2520,2602,2523,2571,2541,2604,2522,2617,2494,2603,2522,2597,2517,2623,2522,2571,2544,4893,1239,1315,1268,1294,596,675,640,633,1249,1347,597,664,599,679,1240,1316,624,680,1257,639,625,680,589,1318,616,650,1283,1296,604,680,1239,1297,1285,647,618,1332,1224,1321,612,689,1244,649,610,688,600,1303,1276,1291,623,672,1240,680,592,669,633,665,616,1290,627,652,1292,619,645,1310,621,642,1249,688,622,1291,613,685,1254,1295,615,683,589,668,1254,1312,1271,659,605,1318,626,660,607,661,1269,1309,591,690,608,664,630,640,647,668,588,661,601,689,625,653,613,666,624,670,608,680,587,690,620,642,1260,657,645,1296,587,689,1247],
  "time": "2023-12-27T12:36:12.180+0100"
}]
```
Received from RTL-SDR :
``
{"time" : "2023-12-27 12:36:11.993760", "model" : "Somfy-RTS", "id" : 1146088, "control" : "Prog (8)", "counter" : 642, "retransmission" : 1, "mic" : "CHECKSUM", "mod" : "ASK", "freq" : 433.492, "rssi" : -0.112, "snr" : 14.872, "noise" : -14.984}
```
=> Does not work shutter does not jog ! BUT fully fonctionnal for UP,MY,DOWN

3. Test from original remote : 
```
[{"encKey": 168,
  "address": 6946444,
  "rcode": 632,
  "command": "Prog",
  "rssi": -66,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [182291,2482,2491,2451,2488,2450,2498,2462,2472,2468,2461,2464,2496,2465,2479,2446,2489,2449,2498,2462,2464,2477,2462,2467,2476,4794,1234,1238,1263,1221,1271,1240,635,591,637,609,615,605,630,599,1270,1229,629,611,1262,1233,1249,1248,625,588,1285,1206,651,596,1271,576,676,584,662,1232,1273,1222,1259,606,659,592,670,597,674,585,659,1232,639,593,1268,585,654,1241,1259,589,669,1230,1262,588,665,1220,653,584,1281,1233,624,593,641,597,650,581,1284,1210,638,599,645,593,1271,1235,612,611,1267,1223,642,596,623,609,628,611,1261,1215,646,592,645,603,627,611,635,584,635,610,613,608,642,606,640,602,626,594,640,600,621,607,632,610,1262,603,659,595,668,1231,1234],
  "time": "2023-12-27T12:56:22.959+0100"
}]
```
From RTL-SDR : 
```
{"time" : "2023-12-27 12:56:22.759786", "model" : "Somfy-RTS", "id" : 9240169, "control" : "Prog (8)", "counter" : 632, "retransmission" : 1, "mic" : "CHECKSUM", "mod" : "ASK", "freq" : 433.491, "rssi" : -0.112, "snr" : 15.175, "noise" : -15.287}
```
=> Work shutter jog !

Do you have any idea ? i'm stuck...

rstrouse, 2023-12-27:
Those frames are identical between your existing remote and ESPSomfy RTS.  While this is a classic 80 bit frame, I wish I could see the pulse timings from the RTL-SDR.  Perhaps try sending a 56-bit prog.

Jezza34000, 2023-12-27:
ok after many and many tests I understood that it is a shutter problem...
 I have reached a stage where even the original remote control no longer responds to the "Prog" command (but the others command do).  I'm currently investigating the manufacturer's documentation to understand what could have caused this m, but at the moment I don't have the answer...

rstrouse, 2023-12-27:
The most common thing that causes this issue is shared addresses between devices.  If the address sends too many commands with an improper rolling code the motor will stop responding to that address.  Perhaps Simu does this only to the prog command.

Typically with Somfy this requires a factory reset of the motor which clears all the addresses from memory and starts over.

Jezza34000, 2023-12-28:
You was right about UP+DOWN @rstrouse it "wakeup" the shutter for programming it...
you should try @ViViDboarder 
I close this issue

ViViDboarder, 2023-12-28:
Is there a way to configure this so Grouping works in the app? Or is that something that could be captured in this issue?