rstrouse/ESPSomfy-RTS #213: Loving the github update feature!

FlaMike, 2023-12-27:
Just wanted to give you a shout out "Thank you!" for adding this feature.  It really simplifies & streamlines updating the programs!  I used it for the first time over the weekend to update to 2.2.  EZ PZ!   I also appreciate showing the version installed on the device's home page.  Nice touches all!

Thanks again for creating and sharing this superb project!  It's an excellent convenience & addition to our home.

rstrouse, 2023-12-28:
If you are paying attention it also has a firmware updater from HA.  This is because I must either be a masochist or I simply need to know the answer to what if.

FlaMike, 2023-12-28:
You are just too to we who enjoy the fruits of your labor!!!  Thanks!  Just installed 2.3 from HA.  So sweet!!!

paede81, 2023-12-29:
Can the git hub updater be used for all esp devices (esp, esp-s2, esp-s3, esp-c3)?

rstrouse, 2023-12-29:
So far yes.

rstrouse, 2023-12-31:
I got around to testing more thoroughly as I only had one sample of a non-WROOM device.  So after v2.2.4 the github updater works reliably with esp32, esp32-s2, esp32-s3, and esp32-c3.   As it turns out that while it initially appeared to update from prior versions it was just updating the littlefs partition.

closed 2023-12-31 by rstrouse