rstrouse/ESPSomfy-RTS #214: Software does not allow for selecting GPIO number above 31

RambunctiousR, 2023-12-29:
Hi, I am using a ESP32-S3 which has 45 GPIOs. The TX pin is on GPIO43 and RX pin is on GPIO44. The software only allows me to select a GPIO up to no.31 so I can't select the correct pin in the radio configure settings within ESPSomfy RTS when setting up. See attached for pin map of the board I am using. I was also going to wire the other pins to GPIO 35-42 to avoid the ones with a touch sensor input as you said you had an issue with wiring to a touch pin.

Thanks in advance for any assistance.

[image]

rstrouse, 2023-12-29:
I will add some code to the UI app to allow for more selections however, the TXD and RXD on other ESP32 devices are not necessarily a good choice.  This is because they are often wired to a USB UART and will cause you all sorts of mind numbing pain.  

I have gotten past the goofball issues with the Touch pins as well by changing the way the pin mode is defined when initializing the receive pin.  I'll post the changes up to the main repo later today.

rstrouse, 2023-12-29:
I have just confirmed that you cannot use pins 43 and 44.  Remapping these pins will cause all sorts of gutteral reactions to the board.  These pins are generally for reading and writing serial data to and from the Serial console.

In general the following GPIOs should be avoided on the S3 `0, 3, 6, 19, 20, 26, 27, 28, 29, 30, 31, 32, 43, 44`.  I am currently working on filtering the list based upon the board type.

RambunctiousR, 2023-12-29:
Ah, thanks for looking at this. Does this mean this board won't work? I'm very new, I can't tell if the TX/RX need to be wired to those specific pins or if they can go to any of the GPIOs. 
I can try and order a new board if this one is no good.

rstrouse, 2023-12-29:
You do not need or want the TXD/RXD pins wired up on the board and your board will work just fine.  These are actually pins that are used to access the flash on the board through a serial adapter.   You need to avoid them.  The TX/RX you are connecting are for the TX/RX pins for the Transceiver.  

These are labeled (GDO0/GDO2) on the transceiver.  I know they made this so blisteringly confusing.  So ESPSomfy RTS will listen to the receiver on the RX pin and send out data through the radio on the TX pin and it can be any of the unreserved pins on the ESP32.   The reserved ones are in the prior post.  These pins that are reserved are for other peripherals that we are not using and can only be used for that purpose.

GDO0 is commonly referred to as the TX pin for the transceiver.  Any signals pulsed on this pin will be sent over the radio to the motors.  ESPSomfy RTS takes care of all the pin control for you after it has been selected into the TX of the radio settings.  

You can also set the RX to the exact same pin and not even hook up the GDO2 pin on the transceiver.  Since the transceiver can only send or receive at any given time, ESPSomfy RTS will multiplex this pin and listen on it when it is not transmitting.  In fact the CC1101 will replicate the receive signals on both pins when in receive mode.  The results will be the same regardless of whether you choose to hook up GDO2 or not.

With the touch pins, I originally though I could just set the output mode for them initially and all would be good but later realized that the pin modes had to disable the touch function which ESPSomfy RTS now does if the pin is mapped.

RambunctiousR, 2023-12-29:
Sorry, I thought the TX/RX on the transceiver need to to match up to the TX(D)/RX(D) on the ESP32. 

I have rewired it and seems to be working well. Many thanks, this is really amazing job.

rstrouse, 2023-12-31:
I have posted the new version to include all the available/usable pins for the S3, S2, and C3.

closed 2023-12-31 by rstrouse