rstrouse/ESPSomfy-RTS #215: Updated to 2.2.4, now no shades available.

Calimerorulez, 2023-12-30:
Just updated one of my controllers to 2.2.4, and now my two shades are missing.

MQTT and Wifi settings are correct.

Calimerorulez, 2023-12-30:
Restored the shades by restoring the specific entry from the generated backup.

rstrouse, 2023-12-30:
Don't update to 2.2.4 for now.  Github requires that I publish the release in order to test it.  The original compile files appear to be missing something so I have to do the Arduino dance.

rstrouse, 2023-12-31:
You can update to 2.2.4 at your leisure.  I did some dance fighting with Mr Arduino and finally figured it all out.  Sometimes it's like weaving in and out of traffic in a full size game of Frogger.

isngofoz, 2024-01-01:
Unfortunately, I decided to be adventurous this morning, and used the GitHub update feature. It was a disaster as I could not get the ESPSomfy RTS to reload. I had go back to basics to reflash the chip again after having forgotten how I did it previously. After many attempts of fussing around with different versions and restoring backups, I finally landed on v2.2.2 working again. 

Then comes another problem, the ESPSomfy UI can no longer control 2 out of 3 shutters. The original remote still can control those 2 shutters. 

Incidentally, those are the 2 shutters that have automation in HA to control them based on occupancy. I have got occupants in my house that have got used to the automations that are asking why the shutters aren't opening now. Any idea what's wrong? Do I need to re-pair the shutters again?

rstrouse, 2024-01-01:
Did you restore the backup that was generated when you updated the firmware?  It will contain all the rolling codes.

isngofoz, 2024-01-01:
Okay, I just did that and the 2 shutters appear to be working again. Thanks heaps for the quick support. 

I didn't think that I should use that backup since it was from v2.2.3 and I am using v2.2.2 version now. I have confirmed that the one of the shutters have worked with the automation again. 

I will monitor the other shutter although I have a feeling that the motor on that shutter is on its way out as it is struggling to open each time. I guess I will need to re-setup that shutter again if the motor gets replaced?

rstrouse, 2024-01-01:
Awesome.  I have worked to make that backup file work forward and backward since 2.1.x.  Sorry about the publish but GitHub requires the repo be published in order for the files to generate.  However, I made it so I can still publish if I need to set the binaries while warning in the updater that the release is still in draft.  v2.2.4 is now valid on the repo.

isngofoz, 2024-01-01:
No apologies necessary. Thank for your ongoing hard work.
I will wait a couple of days before I tinker with another update attempt, else the house occupants will not be too happy with me.

Calimerorulez, 2024-01-01:
Updated fine now to 2.2.4. Shades are retained now :)

Happy New Year and thanks.

closed 2024-01-01 by Calimerorulez