rstrouse/ESPSomfy-RTS #216: Update to 2.2.3 error: ascii codec can't encode character / ordinal not in range

groosman, 2023-12-31:
Thanks for the great work, I'm able to operate all the shutters in my house via this integration! Especially the clear and step by step description helped me a lot.

The update to v2.2.3 automatically appeared but when I push the install button the following error shows up:
> Kan service update/install niet aanroepen. 'ascii codec can't encode character 'Ixe9' in position 635: ordinal not in range(128)

Someone with the same issue? Any advice on where to look for fixing this?

rstrouse, 2023-12-31:
This has to do with HA trying to write the backup file with encoding.  I will change this to ensure it does not try to encode the characters for the Dutch character set when it is trying to write this file in the next release of the plugin (Hopefully today).

groosman, 2023-12-31:
Thanks @rstrouse!

rstrouse, 2023-12-31:
Update the plugin in HA and this should now properly save your backups.

groosman, 2024-01-01:
strange, I first updated the plugin but with no success. Eventually deleted everything and reinstalled the integration. I struggled through some "invalid flow specified" issues but got the integration working again. When running the update I get the following error (even after rebooting & CTRL-F5): 
[image]

rstrouse, 2024-01-01:
That is weird.  I'll take a look.

rstrouse, 2024-01-01:
Ok I guess I needed to actually write a file from a production system instead of my dev system.   Writing the file required first converting it to a byte output before writing.  It is odd that I can't just write an ordinal string.  One would expect to simply write a file and be able to read it back out exactly as it was written.  In v2.3.0 of the plugin that is exactly what it does but I had to transform it from one form then to another.

groosman, 2024-01-01:
hey @rstrouse thanks for the follow-up. Nothing on my side that could cause this?

rstrouse, 2024-01-01:
Nope it just required coercion because of the non-Ascii characters you used to name your shades.  The firmware handles it just fine it was just the Python subsystem in HA that didn't want to write the file because it had non-ascii characters in it.   Then when I told it that it was supposed to ignore it, it blew me off and said that I needed to encode it anyway.  Hence the second message.

The Invalid flow you got after the update is because HA needed to restart completely.  A few releases back they added a quick reload that does not go through the full restart process and initialize the custom components.   As a result, the it doesn't always load all the changes until a restart.  So when you update a HACS component it is supposed to restart but sometimes it does a quick reload.

If you update to v2.3.0 for both the firmware and the HA integration all should be good.

groosman, 2024-01-01:
okay, I'll try that tomorrow or day after tomorrow and update you here! Thanks again for the great work.

groosman, 2024-01-02:
All went well @rstrouse I've downloaded the firmware and application and updated via webinterface. Did a full reboot on HA and it's up-to-date. Keep up the good work!

closed 2024-01-02 by groosman