rstrouse/ESPSomfy-RTS #217: Update to v2.2.4 - all shutters and blinds disappeared

DUC750, 2024-01-01:
After updating to 2.2.4, all my shutters and blinds are gone. All my backup files are shown as invalid. Even after a backflash to 2.2.3.
[image]

rstrouse, 2024-01-01:
Remove the .txt extension from the file.

rstrouse, 2024-01-01:
Upload one of those backup files here.  I thought for a moment that I may have had a mime check on it but that message is a service return.

rstrouse, 2024-01-01:
The story behind this is in this ticket #215.   Changes being made to support other boards required posting interim releases in pre-release but unfortunately they showed up on the OTA.  This all has been rectified but in the meantime it looks like some of the interim releases got installed.

Keep going back in time with the backup files until you find a good one.  The rolling code persistence keeps the data in non-volatile storage on the chip as well and backups are backward compatible.

DUC750, 2024-01-02:
Thanks for the help. I rolled back to 2.2.3, removed the extension .txt from the backup file and successfully imported the backup.

However, I then updated to 2.3.0 and now my ESP is no longer reachable at all. I'm just going to reflash him completely over the next few days. But this is another problem. The issue can be closed.

And please excuse my rudeness, I wish you a good, healthy and prosperous 2024!

Thank you very much, also for this great project. 

Greetings
Mirko

DUC750, 2024-01-03:
Just to complete it, after a complete reflash with 2.3.0 everything is running great again! THX

closed 2024-01-03 by DUC750