rstrouse/ESPSomfy-RTS #218: Is there some way we can contribute/donate/support?

joneshf, 2024-01-01:
@rstrouse I've seen a couple issues/threads where folks have wanted to contribute/donate/support in some way. Each time, you respond with something along the lines of "just glad you found it useful." It's an admirable response. But, is there any thing we can do to show our gratitude? Donate money? https://www.buymeacoffee.com/? Patreon? GitHub Sponsor? Donate hardware? Spread the word? Arbitrary praise? Convince HomeAssistant that 100% is closed, not open 😅? Something else?

It's fine if the response is, "nah, I'm good." But I feel like ESPSomfy-RTS is one of the better put-together projects around:
- It's well documented.
- The reasons for decisions are explained well.
- The UI provides a good UX
    - It's easy to do the right thing.
    - It's hard to do the wrong thing.
    - We're warned asked to confirm before dangerous stuff.
- You're responsive and helpful.
- Perhaps more importantly, you're not condescending (from what I've seen anyway).
- It works better than commercial options.
- It provides more features than both the actual remote and commercial options.
- etc.

And if there's something I can do to show that (besides tell you in this GitHub issue), I'd like to do it.

rstrouse, 2024-01-01:
> But, is there any thing we can do to show our gratitude?

You just did thanks!

> Convince HomeAssistant that 100% is closed, not open 😅?

Is that not the most illogical thing you ever heard?  What is funny is how the response to simply pointing out how silly it is, can be so nasty.  I think it is even weirder when the concept is just explained as a matter of fact.  Like you swallowed the wrong pill and reality is somewhere outside your grasp.  But then again I am pretty sure rabbits don't poop hard boiled eggs.  And if there is a big guy in a red suit standing next to my fireplace in the middle of the night... he's probably there to rob me.

Thanks for the feedback and Happy New Year!

joneshf, 2024-01-01:
> > Convince HomeAssistant that 100% is closed, not open 😅?
> 
> Is that not the most illogical thing you ever heard? What is funny is how the response to simply pointing out how silly it is, can be so nasty. I think it is even weirder when the concept is just explained as a matter of fact.

It definitely seems like a weird hill for them to die on. It's the opposite of how a lot of people think (it's basically a [glass half-empty/half-full situation](https://en.wikipedia.org/wiki/Is_the_glass_half_empty_or_half_full%3F)). And in a case like mine–where I've only automated horizontal blinds with a tilt-only mechanism–percentage is the last thing I would expect. The angle of the blinds is more useful than what percentage of open the "cover" is.

I can't think of any advantage there is to keep this tech debt in Home Assistant, instead of updating the model to reflect a better understanding of what a "cover" is. But then again, I'm not a core contributor, so maybe keeping the tech debt makes sense.

joneshf, 2024-01-01:
> > But, is there any thing we can do to show our gratitude?
> 
> You just did thanks!
> 
> Thanks for the feedback and Happy New Year!

Fair enough! Well thanks again. If you change your mind, and want/need something, let me know.

closed 2024-01-01 by joneshf