rstrouse/ESPSomfy-RTS #221: Can we have longer names?

joneshf, 2024-01-03:
We have a complex house. The naming of windows can get a bit long. For example, I'd like to have a shade called "Main Living Room Blinds 1" or "Next Gen Dining Room Blinds 2". Or a group called "Main Living Room Blinds" or "Next Gen Dining Room Blinds". But all of these names are longer than 20 characters. I _think_ all the names I'd want would fit in 32 characters, but 64 would give more headroom. Is that a realistic change? Or is 20 characters the max we can do?

If having longer names means a single ESPSomfy-RTS device would support fewer shades/groups, I'd rather make more ESPSomfy-RTS devices and have more correct names, than have to truncate the names for more shades on fewer devices.

rstrouse, 2024-01-03:
Strings are my enemy.  They eat up most of the resources on the ESP32.  I am kicking around the possibility of implementing rooms which would reduce the overhead of repeated strings.  This is all still pretty infant but it is probably a better overall solution.

joneshf, 2024-01-03:
Gotcha. Well I dunno how complex you want to make it (nor if it would help with resources) but since it's early on, if we could have them nestable–like "Locations" in [OpenHAB's Semantic Model](https://www.openhab.org/docs/tutorial/model.html)–that'd be ideal, from my perspective.

For posterity, it would let us model things like:
- Main
    - Dining Room
        - Blinds
    - Living Room
        - Blinds 1
        - Blinds 2
- Next Gen
    - Dining Room
        - Blinds 1
        - Blinds 2
    - Living Room
        - Blinds

rstrouse, 2024-01-06:
I am really getting pretty limited on PSP and local memory so I will probably be limiting this to a single leaf.  Still not sure if I push this data to HA and MQTT at this point though.

rstrouse, 2024-01-08:
If you install the pre-release of v2.3.1 it now contains rooms.

rstrouse, 2024-01-22:
v2.3.1 Has been released and it contains the rooms function

closed 2024-01-22 by rstrouse

joneshf, 2024-01-25:
Awesome, I'm excited to try this out!

Re: pushing the data:

> Still not sure if I push this data to HA and MQTT at this point though.

I didn't see anything in the release or wiki about how this will behave, but I might've missed it. Does the data end up getting sent to Home Assistant? For example, if I were to make a room called "Main Living Room" and then put "Blinds" in that room, how would it show up in Home Assistant? Would I get a device called "Main Living Room Blinds"? Would I get a device called "Blinds" in the "Main Living Room" area? Would I get a device called "Blinds" without any area? Or something else?

rstrouse, 2024-01-25:
Actually, none of the above.  You can create all of those scenarios in HA without ESPSomfy RTS setting it up.  Unfortunately, in HA I am unaware of any way to detect whether the user made any changes.   So doing that might make 50% of the folks happy while making 50% sad.  Not creating rooms and re-assigning them can make 100% happy.