rstrouse/ESPSomfy-RTS #222: Login no longer available

Ric57100, 2024-01-04:
I wanted to connect on the UI to install new firmware update but login page is not available.

[image]

HA integration works well, but I can no longer manage anything from the GUI.

rstrouse, 2024-01-04:
Clear your browser cache or perform a refresh procedure if you are using chrome.

Ric57100, 2024-01-04:
> Clear your browser cache or perform a refresh procedure if you are using chrome.

I've already done it. I have the same issue with Firefox, Chrome and Edge.

rstrouse, 2024-01-04:
What firmware version are you currently on?

rstrouse, 2024-01-06:
It looks like the filesystem has been corrupted.   This can happen when another process on the esp32 commits data to the filesystem when it is updating.   I am working on making it so the filesystem is not written to during this process but in the meantime it means that the firmware will need to be updated again.  Please make sure you have a current backup.  To do this you can run the following command.

`http://<ip addresss>/backup`

Then double check to ensure that file is not empty.

Next run the following command and let it complete without moving any of your motors until it has completed.  This may take a few minutes.

`http://<ip address>/downloadFirmware?ver=v2.3.0`

At the end of this it should have refreshed the file system files for you.

rstrouse, 2024-01-16:
@Ric57100 did you get squared away?

Ric57100, 2024-01-16:
Sorry for the late response. Yes, I managed to solve my issue by applying latest firmware from scratch and adding my motors one by one.

closed 2024-01-16 by Ric57100