rstrouse/ESPSomfy-RTS #225: Problems with connecting to Somfy RTS motor

Pleeriezer, 2024-01-05:
I have flashed the latest Firmware to the ESP version 2.3.0

I having problems right now to conncet to a Somfy RTS shade (is is the only one we have), for the update it was working.

Tried to switch te RX an TX reset the motor, reboot the ESP butt stil not working.
Radiosignals seems to be good so what can be wrong?

Thanks and greetings,

Erik.

rstrouse, 2024-01-05:
Are you saying that the UI for ESPSomfy RTS comes up but the radio is not communicating?

Pleeriezer, 2024-01-05:
Hello,

I see the radio signals in the Log but te it is not connecting with the motor?

rstrouse, 2024-01-05:
Which motors do you have?

Pleeriezer, 2024-01-05:
1 Motor original Somfy RTS

rstrouse, 2024-01-06:
Upload your backup file and let me have a look at the configuration.

Pleeriezer, 2024-01-06:
Hello I have attached the file
[ESPSomfyRTS 2024-01-06T09_56_40.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/13848921/ESPSomfyRTS.2024-01-06T09_56_40.zip)

rstrouse, 2024-01-06:
Please set your Tx power to 10.  Your CC1101 may not be capable of a Tx power of 12.

Pleeriezer, 2024-01-06:
Thank you for your response.

Unfortunately that makes no difference, the AB is 30 cm from the ESP and 4 meters from the Somfy motor

Pleeriezer, 2024-01-06:
I set it to 7Dbm and now it is working!!

rstrouse, 2024-01-06:
10 is probably the max for that transceiver and the reduced power probably caught up to the rolling code.  At 12 it was probably shouting so loud that it was noise.  Not all of the CC1101s can yell that loud and many of the motors just see it as a heated argument.   You will find that the CC1101 has way more range than any remote you have even at 7.

closed 2024-01-06 by rstrouse