rstrouse/ESPSomfy-RTS #226: Simu Hz and other comments

Nickduino, 2024-01-06:
1. What an impressive job you've done on the programming and documentation, I'm in awe!
2. I can tell you that ESPSomfy-RTS works with the Simu Hz protocol. I haven't tested it yet because I'm waiting on my CC1101 transceiver but it has been working with my Arduino sketch and Python script (that implement the Somfy RTS protocol) for almost 10 years now, so...
3. You complain a lot about the range of the Somfy remotes and their use of a PCB antenna, so I thought I'd make you jealous of my OEM remotes that use a coil antenna:
[image]
[image]

Now with my questions:
4. The Tx power is entirely independent from the Rx sensitivity, correct? I can have ESPSomfy whisper its commands (to not broadcast them on the other side of the town) but still hear every command from the original remotes?
5. "Groups do not have a defined position and cannot be controlled by setting the position. This is because the timings of each member can and will likely be different." <- It would still be interesting to be able to close all blinds of a group to 50% or something. Maybe ESPSomfy offer this option (even though, in the back-end, it would send two frames for each blind)

rstrouse, 2024-01-06:
@Nickduino I am so glad to see you.  When I started looking for solutions I came across your PI Somfy development.  Glad to hear the Simu uses the same protocol.  Let me know if there are any issues.   The range of the Somfy remotes has me thinking that I need to do some experimentation on the transmitter.  It might be fun to play around with increasing the output power.

As for your questions, yes the Tx power is completely separate from the Rx sensitivity.  So you can whisper the commands and tune the Rx to pick up your remotes.  The group functions do work like you expect in HA.  I didn't implement that algorithm in the ESP32 at this point because there have been other priorities on the plate and it can be a bit wonky with integrated tilt mechanisms.

Nickduino, 2024-01-07:
> I am so glad to see you. When I started looking for solutions I came across your PI Somfy development.

https://github.com/Nickduino/Somfy_Remote would probably have been a better starting point, it has been forked and ported to ESP8266 and ESP32 many many times.

> Let me know if there are any issues.

I will but looking at your doc, I doubt it

Nickduino, 2024-05-15:
I've been using it for 4 months, it's become extremely reliable with the v2.4.2
(And yes I installed the v2.4.3 since)

closed 2024-05-15 by Nickduino