rstrouse/ESPSomfy-RTS #227: Adding Awning And Lights

vacano, 2024-01-07:
Happy New Year and many thanks for this fantastic project! I do need to polish my soldering skills though :)

I have the RTS Remote for my Awning which has 4 channels and with a push of a button you go to the next one, so basically 1 is the awning and 2 is the lights for this respective awning and 3 and 4 are for the second one. My question I tried to add lights as lights but they do not function, but when I add them as a separate awning they do but in HASS they come not as a switch to toggle but as another awning as well as on the Somfy app. Am I doing something wrong ?

Thanks in advance for your help and thanks again for the project!

rstrouse, 2024-01-07:
You should add the light as a 2 button or 1 button toggle dry contact not as another awning.

vacano, 2024-01-08:
Thank you very much for your quick reply. I tried but nothing happens, I have an on/off indication in the app but physically nothing happens.

vacano, 2024-01-09:
Yes this is correct.  Picture of the remote I have.

[image]

rstrouse, 2024-01-09:
Can you capture a log of turning on and off the light from the remote?

rstrouse, 2024-01-16:
@vacano have you had success?  If you haven't please capture the logs by pressing the copy button on the Radio -> Logs tab and paste it here.

rstrouse, 2024-02-04:
@vacano if you are still having issues capture a log using the remote and re-open this issue.  Paste the log results here.

closed 2024-02-04 by rstrouse

vacano, 2024-03-10:
Hello, I was away on a business trip and just came back home. My saga still continues and I cant manage to pair the lights as dry contact as suggested. I somehow managed to it on one of the Awings, but now the second one puts lights off and retracts the awing at the same time ?

ganttview, 2025-02-02:
I think I'm having the same issue.

I can pair the device (lights turn off when I click PROG in the app) but controls from the app don't work. I've tried pairing as dry contact 2 button, 1 button, and roller shade, and have unpaired each time. 

Remote details:
Situo 5 RTS pure remote, group 1 is the awning, group 2 is the lights. Up/down buttons on the remote switch lights on/off, respectively. 

Remote Radio Log (using group 2):

[{"encKey": 174,
  "address": 15096389,
  "rcode": 363,
  "command": "Up",
  "rssi": -22,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [10450,7779,2515,2522,2488,2522,4836,1291,1293,1291,1295,1284,625,642,651,636,1309,1270,1312,620,670,633,635,641,670,621,657,1291,1295,1286,1275,650,660,615,655,641,650,638,659,1275,651,636,660,649,643,629,660,640,1300,1276,1289,647,635,640,653,643,649,636,660,625,667,628,660,1292,643,627,662,639,1300,626,660,1268,1296,1285,1298,1280,1290,649,637,664,626,643,651,636,667,1276,1299,638,641,649,635,1292],
  "time": "2025-02-02T13:12:17.201+1030"
},
{"encKey": 175,
  "address": 15096389,
  "rcode": 364,
  "command": "Down",
  "rssi": -22,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [10457,7767,2524,2518,2525,2493,4835,1284,1292,1271,1315,1271,654,639,667,619,645,633,660,628,665,624,657,639,1318,620,653,642,655,1270,654,635,658,617,670,634,637,653,1291,635,661,622,666,1279,1292,1270,1314,617,669,635,664,1278,645,634,662,625,1295,637,662,1275,647,636,1317,1271,1288,648,643,631,659,641,653,618,675,1266,673,621,1305,614,669,1278,1294,644,654,1264,672,622,657,634,1305,1275,660,623,664,632],
  "time": "2025-02-02T13:12:18.830+1030"
},
{"encKey": 160,
  "address": 15096389,
  "rcode": 365,
  "command": "My",
  "rssi": -21,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [10470,7748,2519,2525,2512,2515,4845,1246,1312,1269,1310,623,667,629,662,643,650,616,678,1266,1296,1283,656,643,1295,1260,665,631,662,642,650,617,1303,1294,643,631,1313,1269,658,640,1306,1271,661,627,1293,1288,649,642,1306,1268,643,632,1313,619,668,1262,676,616,677,614,659,648,1292,1266,1320,1258,677,616,1301,642,648,1269,660,642,647,643,1307,616,670,1271,1310,616,677,614,660,1274,1301],
  "time": "2025-02-02T13:12:20.471+1030"
},
{"encKey": 160,
  "address": 15096389,
  "rcode": 365,
  "command": "My",
  "rssi": -20,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [27641,2520,2520,2507,2526,2500,2533,2494,2515,2519,2518,2520,2519,2500,2531,4833,1264,1301,1299,1288,621,684,610,663,642,653,612,685,1269,1307,1268,668,608,1313,1268,660,635,679,616,651,636,1311,1268,661,617,1327,1276,659,619,1318,1268,642,649,1287,1272,677,612,1309,1268,688,615,1297,631,662,1269,666,630,662,643,650,616,1330,1266,1293,1268,668,622,1310,647,642,1281,655,642,647,637,1312,617,675,1267,1297,630,661,617,676,1268,1314],
  "time": "2025-02-02T13:12:20.872+1030"
}]

Thanks, and great work on this integration and very detailed instructions.

ganttview, 2025-02-02:
One more note - the "My" button also toggles the lights on/off.