rstrouse/ESPSomfy-RTS #228: Won't recieve any button press in Radio -> Logs

KnightsKingdom, 2024-01-07:
As everybody else I want to open/close my sunscreen via your project, from what I've seen it looks amazing.
I've connect my CC1101 radio & the ESP32 and at first glance everything seems fine. The radio is initialized, so far so good.
But when I go to the logs, I don't see any entry after I press any button on my Somfy remote. 
I assume the Tx pin on the CC1101 is responsible for this action, it's connected to the Rx on the ESP32 (GPIO12).
Tried all kinds of different combinations and inputs on the ESP but nothing seems to work, i just can't receive any commands from the remote,
Do you have any suggestion(s) how to tackle this?

Hardware used:
CC1101 -> https://www.amazon.nl/gp/product/B07YX92NMP
ESP32 -> https://www.amazon.nl/dp/B0BS6R68H6

rstrouse, 2024-01-07:
Which Somfy remote do you have?  Please verify that this is not a IO remote and is an RTS.

KnightsKingdom, 2024-01-07:
On the back of the remote it says Somfy 74300. Google learns that it's an (older) type off Somfy RTS.

rstrouse, 2024-01-07:
That is a Telis remote and it should work well.   The hardware you selected should also do the trick.  For a radio comms like this (it is not a serial connection) you should not swap the TX to the RX.  The TX pin should match the GDO0 pin on the transceiver and the RX pin should match the GDO2 pin on the transceiver.  It is never a good idea to use the TXD, TX0, RXD, or RX0 labeled pins on any ESP32 if you have them.  These are serial UARTS that are likely in use for USB or they are attached to a JTAG for debugging.

esalgado, 2024-01-08:
Hello, I tried my remote today after building 2 devices and testing sending from one and receiving from another, it worked very well!

But when testing with the real remote at my friend's house, with a Tellis, it did not work. I downgraded to 2.2.3(was on 2.3.0), and it picked it right away!! Working very well.

KnightsKingdom, 2024-01-08:
Thanks both for the replies but I really can't get it to listen to my remote. 
I now reflashed with 2.2.3 hoping it would solve my issue but again not button presses received although radio is initialized.
It's driving me crazy. Is there anything I can do/check to see why it doesn't work @rstrouse ? Want me to make some pictures of the connections ?

rstrouse, 2024-01-08:
> Want me to make some pictures of the connections ?

Yes

KnightsKingdom, 2024-01-09:
Let's see it this helps:
Below my connection schema, as you can see MOSI is connected to MISO and MISO is connected to MOSI
that way the radio is initialized, the other way miso -> miso and mosi -> mosi does not work 

ESP32	Radio	Color

3.3V	        pin 2	Red
GND	pin 1 	Brown
GPIO5	pin 8	Purple
GPIO18	pin 4	Blue
GPIO19	pin 3	Grey  
GPIO23	pin 5	Orange
GPIO13	pin 6	Green
GPIO12	pin7	        Yellow
[image]
[image]
[image]
[image]
[image]
[image]

https://github.com/rstrouse/ESPSomfy-RTS/assets/54189854/09d452f8-6a43-4c93-a2d8-0b987f18dc55

rstrouse, 2024-01-09:
I took a look at your pics and you only have the GND and 3v3 pins connected correctly.  Unfortunately, I think you fell prey to the crappy way the transceiver is printed.  If you leave the wires connected to the transceiver the way have have them move the wires to the specified pin as I have indicated below.

```
1 - Brown - GND - OK
2 - Red - 3v3 - OK
3 - Gray - GDO0 TX - currently GPIO19 move to GPIO13
4 - Blue - CSN - currently GPIO18 move to GPIO5
5 - Orange - SCK - currently GPIO23 move to GPIO18
6 - Green - MOSI - currently GPIO13  move to GPIO23
7 - Yellow - MISO - currently GPIO12 move to GPIO19
8 - Purple - GDO2 RX - currently GPIO5 move to GPIO12
```

KnightsKingdom, 2024-01-12:
Again, thanks for all the effort you put in to help but unfortunately it still does not work. The radio does not even initialize and in my first setup it was.
Is there anything I can still do? Do you have a link for a proven working radio so I get buy a new one, just to find out if mine is maybe faulty?

esalgado, 2024-01-12:
Hello, I have this one: https://a.aliexpress.com/_EwltBKz

Also this one:

https://a.aliexpress.com/_EyySxIN

Both work, it did not work for me when I selected the wrong wiring, but once I selected the right wiring for the boards (E07-M1101D), they worked.

I hope those ones help you!

Edgar Salgado

rstrouse, 2024-01-12:
Did you make the wiring changes as I suggested if you did, please make sure you set the pin configuration as I show in the screenshot below.  Click the enable radio and press Save Radio.  The hardware you have should work just fine unless it is defective.

[image]

KnightsKingdom, 2024-01-15:
Finally had some time to play around again and compared your last screen with my setup. Apparently MOSI and MISO were switched. Now I'm seeing the responses from my remote!!
Next phase is configuring the setup. Thanks for the patience and the help.

rstrouse, 2024-01-16:
Awesome.  I am going to close out this issue for now.

closed 2024-01-16 by rstrouse