rstrouse/ESPSomfy-RTS #230: AP not showing up on Wifi

Darozas, 2024-01-08:
Hi All,

I have installed latest version but "ESPSomfy RTS" not coming up on WIFI stuck.

Can any one direct me to solution please.

PS reflashed 3 times  using Home assistant ESPhome app

Regards

rstrouse, 2024-01-08:
Have you flashed with the proper onboard firmware for your board.   Also check the logs to see if the ESP is booting properly.

Darozas, 2024-01-08:
> Have you flashed with the proper onboard firmware for your board. Also check the logs to see if the ESP is booting properly.

I have flashed with this "SomfyController.ino.esp32.bin" and im using esp32 devkit v1. Tried 2 bords same results.

Logs are wild with this message

[image]

Regards

rstrouse, 2024-01-08:
That is what you get when you flash the wrong onboard file for the chip you are using.

For instance `SomfyController.onboard.esp32s3.bin` to a WROOM device or the other way around.

rstrouse, 2024-01-08:
I just re-read you previous message you need to download and unzip the SomfyController.onboard.esp32.bin.zip file and flash with the bin file in that archive.

Darozas, 2024-01-09:
> I just re-read you previous message you need to download and unzip the SomfyController.onboard.esp32.bin.zip file and flash with the bin file in that archive.

Thanks it done a trick!!!! Thanks for your great work!!!

Can you pair it without remote as i not planing to use remote on blinds home assistant only?

Regards

rstrouse, 2024-01-09:
Yes you can but you will need to use the virtual remote to simulate an existing remote to set the end limits and prepare it for paring.  If you have a remote you should just use it though as it really makes the process easier.  ESPSomfy RTS will also listen for that remote if the motor receives commands from it.

Darozas, 2024-01-09:
Thanks for your time problem is i dont have one.

Is there a way to check if radio is working?

Sorry for to many questions

Regards

rstrouse, 2024-01-09:
What were you controlling them with before?

rstrouse, 2024-01-09:
Ok so set up the motor in the software then follow the instructions for the motor to pair the initial remote and set the endstops.  Use the virtual remote tab to send the commands.

Darozas, 2024-01-09:
Never had them before.

Got this as project so got loads of parts but nothing together yet :D

closed 2024-01-10 by Darozas