rstrouse/ESPSomfy-RTS #231: SIMU pairing impossible

schumissimo63, 2024-01-10:
Good morning,
I have SIMU brand shutters, espsomfyrts is correctly configured, I was able to scan all my shutters and read the data specific to each one, but by pressing the "prog" mode on the remote control it is impossible for me to pair them , I have an error message like this "500 service error service: PUT /addshade Out of Memory parsing JSON". I should point out that I have the latest version of espsomfyrts and that I also read topic #212 which deals with this problem. Thanking you in advance for your help.

Jezza34000, 2024-01-10:
Try UP+DOWN instead of PROG it should work to enter in programmation mode

schumissimo63, 2024-01-10:
Thank you for your response, I pressed UP+DOWN, the shutter went down then went back up to mean that it was in PROG mode but it was impossible to pair. I am attaching some photos, am I correct in the settings, please?
[image]
[image]
[image]

Jezza34000, 2024-01-10:
All the commands that we can see in your logs come from the original remote control of your shutters, do we agree?
Because if this is the case your configuration is not good, you cannot add a virtual "remote control" to ESPSomfy RTS which has the same address.

In "Remote Address" you can re use 196607 this addresse is alreday in use by your physical remote, put something else randomly and after use "Add Shade" and Pair it

schumissimo63, 2024-01-10:
yes, that's right, the logs come from the wall switches which control the shutters. I just tried to pair by putting a random number (163400) but that doesn't work either. On the other hand, I managed to program the shutter using the "group" tab.
[image]
[image]

schumissimo63, 2024-01-10:
I installed version 2.2.1 and successfully did the pairing. Could this be a bug?

rstrouse, 2024-01-10:
> I installed version 2.2.1 and successfully did the pairing. Could this be a bug?

2.3.1 is a pre-release and has many updates as I work through finishing off features.  This pre-release likely had partial functionality for Rooms in it so yes it is likely a bug but looking at the code, the room data has been worked out already.

schumissimo63, 2024-01-10:
@rstrouse I wanted to thank you for this excellent work, thank you very much.

closed 2024-01-11 by schumissimo63