rstrouse/ESPSomfy-RTS #232: Radio Not Initialized

Darozas, 2024-01-10:
Hi All 

Can any one point me to right direction please.

Got all wiring correct checked 10 times but still get this "Radio Not Initialized"

Is there any other ways of checking this?

[image]

[image]
[image]

[image]

Thanks in advance.

Regards

Darozas, 2024-01-10:
I have iven installed ESPSompfy on known working module as i still got it soldered to board changed pinout to match and no go still. I know this module works in this setup as i been using it on water meter.

rstrouse, 2024-01-10:
That is not an E07-M1101D you will need to use the the mapping that lines up with the D-SUN.  However, it looks like they also swapped the GDO0 (TX) and GDO2 (RX) pins on that version.  I love the board but if you used the E07-M1101D mapping then it is not correct.  Please note that the DSUN is also not quite right but you can make it work by simply swapping the pins if the board is wired to the E07-M1101D.

If the wiring for the board is set up for the E07-M1101D then your pin settings should be as follows.  Essentially you need to trace the pin function to the setting in the software.  The reason it is giving you a Radio not Initialized is because the SPI pins are not correct.
[image]

Darozas, 2024-01-10:
I Have Used this for pinout.

Ill try to change to see if it starts work.

PCB files will be yours once proven to work so than you can upload on your git 

[image]

[image]

rstrouse, 2024-01-10:
Yep that is the pinout for the E07-M1101D.  Your transceiver remapped the pins to different pinouts so pin 4 on the E07 is CSN but on your Transceiver it is SCK.  There is a footnote on the table to indicate that but fortunately some remapping on the software should solve the issue.

Darozas, 2024-01-10:
I still dont get it as pints on ESP corresponds to once on CC1101 

Mosi 23
Miso19
CSK 18
CSN 5 

But not working.

I have puled out ESP that i have used before with module soldered that been working and eve this board giving me headache

[image]

[image]

Regards

Darozas, 2024-01-10:
> That is not an E07-M1101D you will need to use the the mapping that lines up with the D-SUN. However, it looks like they also swapped the GDO0 (TX) and GDO2 (RX) pins on that version. I love the [...]

I Have Noticed you have "Enable Radio" ticked is that should be ticked?

rstrouse, 2024-01-10:
Yes it should be ticked or it will not initialize and enable the radio.

Darozas, 2024-01-12:
Hi Thanks "Message Gone" 

How to test if its working? i been using virtual remote but not getting any logs. Is logs only recorded using Somfy Remote ?

Regards

rstrouse, 2024-01-12:
That is a receive only log since transmit output doesn't mean anything.  If the radio not initialized message is gone it is communicating over the SPI interface.  Adding this into the logs would just be noise.  If you had two devices you would see the other one receiving.

Darozas, 2024-01-12:
> Yes it should be ticked or it will not initialize and enable the radio.

Ok Thanks.

Got strange issue One module sends commands another one receives .

But wont work other way ? So the one that receives wont send or the one that sends wont receive ?

Is this normal?

Regards

Receiving end 

[image]
[image]

Transmitting end 

[image]

rstrouse, 2024-01-12:
Double check which pin you have connected to GDO0 and GDO2 for the configuration that you have labeled as the transmitting end.  Change the configuration on that one to GPIO13 for both TX and RX.  In that configuration ESPSomfy RTS will multiplex GDO0 to send and receive on the same pin.

Darozas, 2024-01-12:
This one sends 

[image]

This one receives 

[image]

Keep Plying with TX RX on both but cant send message back to sender 

Regards

Darozas, 2024-01-12:
Motor magickly started to work but still get no logs on Sender

Sender controlling motor

rstrouse, 2024-01-12:
The sender will not place a frame in the log.  Only a receiver will.

Darozas, 2024-01-12:
So why receiver receives but wont send?

I tried TX RX
          RX TX
          TX TX 

But no joy

Regards

rstrouse, 2024-01-12:
The transceiver can only send on the pin marked GDO0.  You should trace this wire from the GDO0 (pin 3) on the transceiver to the pin on the esp32 that you have the this connected to.  Then ensure the that pin is the pin you have selected in the TX dropdown.

Darozas, 2024-01-17:
Unit working.

I have purchased Somfly remote and its loging.

Thanks for your superb work.

[image]

Regards