rstrouse/ESPSomfy-RTS #233: Timeout during group creation

schumissimo63, 2024-01-11:
Good morning,
Thank you again for your fabulous work.
I come to you to ask for an improvement, would it be possible to add a delay in sending commands when creating a group. Let me explain, I created a "balcony shutter" group, this is made up of 3 shutters, unfortunately only two open or close, because the third shutter does not receive the command, I had this problem previously with an arduino, but I was able to add a tempo between each command sent. Or is there a way with your espsomfyrts to get around this problem?
Thanking you in advance.

rstrouse, 2024-01-11:
If you created a group in ESPSomyRTS then it does not send a different frame to each of the members of the group.  It sends a broadcast of the same frame so delaying would have no impact.  All 3 shutters are hearing the same frame and all at the same time.  The first suggestion would be to increase the number of repeats for the group.  ESPSomfy RTS already handles the interframe silence between these repeated frames.

Next investigate what is between the motor and ESPSomfy RTS.  If it is stucco then degradation of the radiated signal can be profound.  Most often this will be intermittent if there are signal problems.

Are you sure that the 3rd shutter is actually a member of the group?  There is no bi-directional communication between ESPSomfy RTS and the motor.  That is why it relies on visual confirmation from you when creating the group.  Sadly you will get the same confirmation response whether the motor is opening its memory, closing its memory, pairing, or unpairing.  If it is out of sequence the act of pairing can actually be unpairing.

Finally, many (I have never experienced one that doesn't) motors will respond with a jog as if they successfully paired with a remote even if their memory is full.  This is very likely if you have used other solutions and have not unpaired them from the motor.   In this instance you should factory reset this motor and pair it back up again.

schumissimo63, 2024-01-11:
Thank you for your response, I had a second group of 3 sections, I just tested it and it works correctly, so I deleted the first one and did it again, and now it works. Sorry for inconvenience. You do a great job ;)

closed 2024-01-11 by schumissimo63