rstrouse/ESPSomfy-RTS #235: Home Assistant raises an Update notice but does not progress further in Install

isngofoz, 2024-01-12:
Home Assistant indicates that there is an update but when I click Install, nothing happens. 
[image]

[image]

Background is that my Somfy motor was almost dead so I unpair a shutter today, so that a technician can replace the motor. After that, without removing the shutter instance or configuration, I was able pair it with the new motor, setup the new up and down time limits and save the shade. 

Thankfully all that was relatively easy and stress free and everything worked again. I was dreading the need to remove the shutter configuration and start a new config. 

But it seems Home Assistant thinks that something has updated but it does not allow an Install.

rstrouse, 2024-01-12:
Which version of the firmware are you running?  If you are running a version prior to v2.2.0 then the ability to remotely update the firmware is not yet available on the device and the install button will not do anything in HA.  

Update to v2.3.0 by downloading the SomfyController.ino.esp32.bin and SomfyController.littlefs.bin files to your computer and update your device.

isngofoz, 2024-01-13:
It was already on v2.3.0 and I have updated using the method you proposed above. Didn't know that it was possible to update the firmware using the HA Install too.

Anyways, weirdly about a few hours when I came back to the computer, the above Update message disappeared.

isngofoz, 2024-01-14:
Thanks. Probably should have waited for a few hours.

closed 2024-01-14 by isngofoz