rstrouse/ESPSomfy-RTS #236: Pinout for Wemos S2 mini

nardev, 2024-01-13:
I see you have a build for this one but i can't find proper wiring.

Any help?

[image]

rstrouse, 2024-01-13:
I have not build an S2 but based on the peripheral descriptions for the pins I would avoid the following pins 0, 3, 15, 16, 17, 18, 33, 34, 35, 36, 37, 38, 39, 40.  Pins 0 and 3 are bootstrapping so avoid the trouble.   Pins 33 - 38 are for the octal flash so just so the manufacturer didn't accost them for multiple versions I just avoid them.   Pins 39 and 40 are JTAG pins so avoid those.

So as a guess use the following but really any pin for any purpose except the above pins should be fine.  You could also used 33-38 if the board doesn't monkey with the SPI flash.

Pin | Description | ESP Pin
-- | -- | --
1 | GND | GND
2 | VCC | 3v3
3 | GDO0 - This is the TX Pin | GPIO 4
4 | CSN | GPIO 5
5 | SCK | GPIO 7
6 | MOSI | GPIO 9
7 | MISO | GPIO 11
8 | GDO2 - This is the RX pin | GPIO 6

rstrouse, 2024-01-14:
Let me know when you have it working and I will include these recommendations for the S2.

nardev, 2024-01-14:
This was the reason why i needed this.

[image]

[image]

rstrouse, 2024-01-14:
Awesome!

nardev, 2024-01-14:
I figured, that would be the smallest form factor and also USB-C option. With shipping it's like 5$ for several PCBs. If you want, i can send you some.

rstrouse, 2024-01-14:
I bought a Seeed Studio Xiao S3 and C3.  These are really small but I am most intrigued by the external antenna decal.

https://www.amazon.com/dp/B0B94JZ2YF?psc=1&ref=ppx_yo2ov_dt_b_product_details

paede81, 2024-01-17:
I have an S2 and this is my GPIO configuration:

[image]

rstrouse, 2024-01-17:
Thanks I was unsure of the DAC pin 18.

nardev, 2024-01-17:
Already ordered PCB, would be able to set it the way in the table up and let it work?

rstrouse, 2024-01-17:
It should work with either configuration.  In the end we can add the boards to the config to auto select the pinouts.  The system does this now based on the chip model.

nardev, 2024-02-09:
Just tried uploading ESP32-S2 version to "Wemos S2-Mini", it writes the fw ok but AP never gets up.

Other than that, any other fw/code works ok on the board, can you please check pins?

It's same board like this one: https://xyzdims.com/3d-printers/misc-hardware-notes/iot-wemos-s2-mini-esp32-s2/

Anything sensitive in code that puts board in other modes?

nardev, 2024-02-09:
@rstrouse sorry, on the other thread i was actually talking about this one.

nardev, 2024-02-09:
@rstrouse 

This is what i get:

```
$ esptool.py --port /dev/ttyACM0 flash_id
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 84:fc:e6:cf:4a:ba
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Flash type set in eFuse: quad (4 data lines)
```

The wifi AP doesn't get created at all.

rstrouse, 2024-02-09:
Unplug the transceiver and see if it starts.  However, the unsupported detection protocol seems to point to the wrong board type for the onboard file.

nardev, 2024-02-09:
Same thing just happened on the other board which was working (ESP32 chip) but as soon as i connected the module, it started ok.

nardev, 2024-02-09:
I'll try the same with wthi ESP32 S2/2S

rstrouse, 2024-02-09:
That is the one chip I do not have is an S2 but I do have a few S3s.  The S2 I ordered has been cancelled twice so I will try again.   What are you flashing with?  Did you pull the zip file from the release?

The other thing that is weird is that it is showing FNR2 which if I read the datasheet correctly is only 2mb of flash.   That wont work if it is correct because the firmware alone is 1.2mb and the filesystem is 1.4mb.  It is possible that the esptool version does not have the correct board definition.

nardev, 2024-02-09:
Can you reach me, i'll send you a few. vedran@nardev.org

rstrouse, 2024-02-09:
I already ordered 4 just now.

One other thing is that it looks like from the screenshot it is not returning all of the data.   Check to see if there is a Javascript error in the console.  It will be something like an cannot read maxXXX property of null.   If that is the case flash the pre-release of v2.4.0 using the onboard file.   This has some trimming on the json output.

nardev, 2024-02-09:
It was my fault somewhere. I have everything up but i can't restore the backup...

It does restore ip settings but remote codes and timings, it just skips it :(

nardev, 2024-02-09:
Ou, i got everything back after the reboot. COOL

closed 2024-02-09 by nardev

nardev, 2024-02-09:
Just to confirm that the wiring schema is correct.

cook1emr, 2024-12-09:
> Pin | Description | ESP Pin
> -- | -- | --
> 1 | GND | GND
> 2 | VCC | 3v3
> 3 | GDO0 - This is the TX Pin | GPIO 4
> 4 | CSN | GPIO 5
> 5 | SCK | GPIO 7
> 6 | MOSI | GPIO 9
> 7 | MISO | GPIO 11
> 8 | GDO2 - This is the RX pin | GPIO 6

I have tried the above pin mapping, however am finding that blinds/garage doors are not very responsive compared to the esp32 wroom variants. Has anyone had success with other pin mapping.