rstrouse/ESPSomfy-RTS #237: Controlling blinds with integrated tilt

perfrankling, 2024-01-13:
I have blinds controlled by Somfy Telis 4 remotes. To tilt the blinds, the remote sends "Step Up" and "Step down" commands. Following an "Up" movement, the tilt will be at 0% and following a down movement, the tilt will be at 100%.

When looking at the ESPSomfy RTS home screen and controlling the blinds with the original remote, the % tilt and display of the position of the blind is correct.

However, when trying to control a blind from ESPSomfy RTS, I cannot get it to behave the way I expect. The "Tilt type" is set to "Integrated".

1. In the Virtual Remote, the Step Up and Step Down buttons have no effect
2. On the home screen, when I choose  (for example) "40%" target position", with a "50%" tilt, with the starting blind position being 0% (at the top), it will move to 40%, briefly stop, but then seems to move to the "My" position, which for me is 100% with 75% tilt.

As far as I know, there is a "Stop" command in the RTS protocol - at least my Somfy RS485 RTS box can send it. It would be great to be able to send it from ESPSomfy RTS.

perfrankling, 2024-01-13:
This is my remote and what it sends.

To stop the blinds, I would need to use StepUp or StepDown, if there is no separate "Stop" command. Sending "My", you never know whether the blinds stop or move to the "My" position.

[image]
[image]
[image]

rstrouse, 2024-01-13:
Yep that is a modulis remote.  Set the bit length to 80 bits for the motor and it will send the same step up/down commands from the virtual remote.

perfrankling, 2024-01-13:
That was it! I should have seen the 80 bits in the log already.

How do I store the "My" position, so that the ESOSomfy does not lose track of the blind when I press "My" on the remote?

rstrouse, 2024-01-15:
You are welcome.  Closing this issue as I believe you are squared away.

closed 2024-01-15 by rstrouse