rstrouse/ESPSomfy-RTS #240: flash read error 1000

effendieffendi, 2024-01-15:
Hi, I'm trying to upload somfyrts to esp-wroom 32. I uploaded SomfyController.ino.esp32.bin using Web-esphome. When you try to start it, you get the message: 
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
 flash read err, 1000
ets_main.c 371 `

Please help me get started

rstrouse, 2024-01-15:
You are using the wrong file.  Download and unzip the SomfyController.onboard.esp32.bin.zip file.  Extract its contents and flash with the SomfyController.onboard.esp32.bin file in that archive.

effendieffendi, 2024-01-15:
Thanks @rstrouse for the reply. I did as you advised. Unfortunately, it still doesn't work. I noticed that the module boots without pin12 connected. But then, of course, the radio doesn't work.
[https://ibb.co/3MtZjCP](url) CDSENET E07-M1101D-SMA

rstrouse, 2024-01-15:
Please double check all of your connections especially the 3v3 and ground connections.  Make sure you have not connected to the 5v pins on the ESP32 and you have not grounded any of the pins besides the GND pin on pin 1 of the transceiver.

rstrouse, 2024-02-04:
@effendieffendi I assume you got your issues hammered out with the wiring.  If you didn't please start another issue as this one is stale.

closed 2024-02-04 by rstrouse