rstrouse/ESPSomfy-RTS #241: MQTT Auto Discovery not available after restart of client

xvisor16, 2024-01-15:
After an MQTT-AD client is restarted the ESPSomfy-RTS stays uncontrollable until a reboot of ESPSomfy.
This means that every time you need to completely restart ESPSomfy after an update of the MQTT client software.
Other solutions like Zigbee2mqtt or ZwaveJSUImqtt don't need that they are available after just a few seconds after restart.
Probably this has to do with those tools continuously sending data out to MQTT.
Is at possible to have ESPSomfy send a discovery/status packet or something every minute when the Publish Discovery box is ticked?

rstrouse, 2024-01-15:
I'll look at what I can do.

rstrouse, 2024-01-15:
In v2.3.1 the software will persist the disco topics and drop them when the reference is removed.  This should allow the subscriber to catch these whenever they subscribe.

xvisor16, 2024-01-16:
Hi @rstrouse,
I've updated to 2.3.1-pre. But this didn't help. The same effect occurs.

rstrouse, 2024-01-16:
Install Main.  I haven't had time to rebuild the pre-release yet with the new commit.

xvisor16, 2024-01-16:
Installed Main and the problem is solved! Thanks 🎉

closed 2024-01-16 by xvisor16