rstrouse/ESPSomfy-RTS #243: Pairing issues

danw87, 2024-01-16:
Hi,
First off, fantastic project, thank you very much!

I'm using a E07-M1101D with ESP Wroom 32, RTS blinds, radio is initialised, read remotes done and on the 2 blinds I did pair (after lots of trying) I can send commands fine too.

For troubleshooting I tried using the remote ID and rolling code from a working physical remote and the commands work fine, issue appears to be entirely related to pairing.

I saw some posts about RX power so have tried from 0 to + 10, to get the 2 to pair that have worked I held the prog Hutton on the virtual remote button for an extended time, also have tried 6 and 20 times on repeat commands. Frequency set to 433.42 Mhz, remotes are detected as 433.33 though

Have tried holding next to the blinds as well as in another room but still struggling.

Initially I couldn't get any to pair (hadn't tried cloning from a remote so don't know if it was TX or pair issue) so I ordered another c1101 and have since managed to pair 2.

Any suggestions? I've got a bunch more devices to add. Didn't have any pairing issues on my own code on FS1000a transmitter but this solution is much nicer 

Thanks!

rstrouse, 2024-01-16:
While this may not be your situation I have seen many folks that are coming from different platforms.  So here are some things that are general rules when it comes to these motors and remotes.

1. Under no circumstances should you ever share an Address with an existing remote.  The problem with this is that the motors and remotes have internal counters for the rolling code which will eventually get out of sync.  When this happens you won't be able to talk to said motor on that address anymore.  There is a window for the code and sometimes you can issue several commands in a row so the remote catches up with the motor's rolling code.  However, after many tries the address will eventually become abandoned in the motor's memory.
2. Whenever a remote is paired and no longer in use it should, (bordering on must) be unpaired.  Unfortunately, there are a limited number of address slots on the motor.   While the general thought is that there are 12 for most Somfy motors this doesn't always ring true.  If these slots become full, the motor will continue to work with the existing remote but simply refuse to pair with another.  Given the description above this condition seems likely on your motors that will not pair.
3. Never assume you are the only one that paired a remote to the motor so there must be many slots left.  Sadly, installers often pair up known working remotes and never unpair them.  
4. If you have been trying to use some of the other integrations (or even this one) you may have been inclined to pair an address without un-pairing it.  Sometimes you end up fiddling so much that you fill up this memory.
5. Finally, with some battery powered motors their pairing ability seems to fall off if the battery is not fully charged.  I can't put a finger on why but perhaps it has to remain awake longer while pairing.

So what does all this mean?  It simply means that once the rolling codes and abandoned addresses fill up the slots you must factory reset the motor to clear them out.  I have seen this many times and in fact that is what Somfy recommends (without saying what causes it) when you cannot get a remote to pair.  This has been true with one of my own motors where I swear there were many slots left.  I have unpaired and paired this same motor with a variety of devices since countless times.

mensand, 2024-01-23:
HI i have a issue with binding. I have the motor react to the prog command of the ESPSomfy RTS. 
So first is put the motor in pairing mode with the real remote en then press the prog button on the config screen. The motor jiggles and the binding is complete. But i can't send command to it. Nor up down or my is working. Do you have a hint. Or should i also go through a full reset ?

rstrouse, 2024-01-23:
> HI i have a issue with binding. I have the motor react to the prog command of the ESPSomfy RTS. So first is put the motor in pairing mode with the real remote en then press the prog button on the [...]

Yes I would factory reset that motor.  Look online for your motor to get the right procedure to do so.

DaanVanDB, 2024-01-30:
Hi, Also issues here with pairing. One blind worked fine (is on a remote with single channel). Then I have 3 blinds on a multichannel. The first blind paired just fine. The other two I just can't get paired. I removed al existing pairings from the motors. Added the multichannel again (which worked fine) but when trying to pair with the ESPSomfy again nothing happens after clicking the pair button. Using ESP32Wroom and E07-M1101D-SMA. Thoughts?

rstrouse, 2024-01-30:
@DaanVanDB 

It sounds like you just occupied the pairing slots again after removing them.  The issue is that in order for the motor to clear paired remotes from its memory it must be factory reset or the remote address must be available to respond to the unpairing.  Sadly, this happens any time a remote is paired and not properly unpaired and this can happen long before the motor ever gets installed.  It can also happen when you look for a solution to control your blinds and pair multiple addresses to the same motor.  

Another scenario is when the address gets used and paired at one time to the motor but the rolling codes are out-of-sync.  A motor that receives a prog command with a rolling code it doesn't like will reject it.  In this instance it simply does nothing.  Unfortunately, there are a bunch of integrations out there that suggest scraping the address from the remote without going through the proper pairing/unpairing procedure.  This results in simply occupying abandoned address slots in the motor.

Also some (a few brands) motors require you to press and hold the pair button until the motor responds.

The go to answer is almost always factory resetting the motor and then pairing up the remotes if you are sure the ESPSomfy RTS device is in range.

danw87, 2024-01-30:
It's the same issue I'm having and I can guarantee you it's not the motors being full. It's happening with most of mine, over 10, including a bunch that are factory fresh and ones that I developed some old code on which I can pair and unpaid other remote fine leaving a space for this remote to pair. I'll fire up my SDR over the next few days and compare the programme command between ESP Somfy to a working remote remote

danw87, 2024-02-05:
I had a look for my SDR but it's not where I though it was after a house move, I'll have a dig through the loft and hopeifind it at the weekend. 

In the meantime while I lost my existing shades I had a play and managed to get all of my shades to pair with my Arduino sketch with no hassle so issues definitely related to the c1101 or something in this code, SDR should hopefully pinpoint what it is once I can unpack it

DaanVanDB, 2024-02-05:
Sorry for the late update. Didn't have time to have a look before. But tried again and think they problem was related to the remote address. I deleted a shade before it was paired. Created a new one and automatically the same address was used. Didn't notice this at first but once I changed the address I could pair the other two shades without any problem. No idea why it didn't work with the other address. For the rest it works like a charm, thanks @rstrouse !

rstrouse, 2024-02-07:
@danw87 when you are doing your testing make sure you set the TX power back to 10dBm or lower.  For the E07-M1101D anything higher than that will likely get transmitted as gibberish.  I don't think the oscillator can transmit higher than that.  I have an SDR (my first one) coming tomorrow because I don't have enough gadgets laying around.

There is something that I forgot.  Are the blinds actual Somfy blinds or are they some other manufacturer.  I have seen a couple of these that the pairing is done with either the my or the down button.  You can try that by using the remote to put it into pairing mode then going to the Virtual remote and pressing the buttons there.

RomanVolcic, 2024-02-12:
Dear rstrouse, first of all THANK you very much, you did a really great job. I put all hardware and basic functionality in one day. I used exactly the same parts as you. Most of functionalities started immediately. I have two problems. First one is pairing. My ESP-RST worked from scratch  even without pairing. I can control my five Somfy blinds the same way as with Telis 4.  When I use Telis after that I need to press buttons few times (sometimes more) to start working again. It is the same with ESP-RST. I assumed that is because Telis and ESP-RST are not synchronised but from above I now understand that this is pairing between motor and remote. 

So I tried to do pairing but I don't know what exactly is pairing functionality and how to know that pairing is successful. I thougt that pairing will synchronize Telis and ESP-RST.

So generally ESP-RST is working but using both of them is not convenience at all.

Do you have any suggestion? Thanks, Roman

rstrouse, 2024-02-12:
ESPSomfy RTS generates a random start number from its mac address.   Most often this value will not coincide with the address of your remote.  It is very likely that if you never changed the address that ESPSomfy RTS came up with it would not match your remote.

rstrouse, 2024-02-24:
@danw87 I assume your issue has been resolved.  If you are having further issues pleas submit a fresh one.

closed 2024-02-24 by rstrouse

ThibautMeric, 2024-08-18:
Hi @rstrouse,

Thank you so much for this project. It seems to me that a huge amount of evenings and coffee went into it ^^".

I'm writting this today because I'm working on my second implementation of ESPSomfy. (the two devices are not  on the same location)

Both devices share the same hardware (ESP32-WROOM-32D & E07-M1101D). My first setup went pretty well, I'm now able to drive 5 shades from anywhere 🎉.

On my second device, I must admit that I'm completly stuck.
The goal is to drive a Dexxo Pro 800 RTS (which is a garage door).
I have spent a couple of hours trying to make the pairing work but it doesn't want to happen :(

What I know is:
- My version of ESSomfy is 2.4.7
- I have the radio working fine as i can scan network and detect button presses on a "smoove origin RTS" remote. 
- From log reading, the protocol is RTS 56Bits
- I can link the physical remote to my logical device in the app
- Motor is on pairing mode when I try to add the device
- Base frequency seems (to me) to be "unsatable". Sometimes the scanning recommends 433.36Mhz sometimes 433.42Mhz ( other values can also be returned)

Did you ever experienced issues with this kind of device? Or am I doing something wrong?

Regards,
Thibaut

Note: I've kept the default remote address, as recommended and didn't touch at the rolling code to avoid messing with the system ^^"

rstrouse, 2024-08-18:
First, lets make sure your ESPSomfy RTS device is transmitting.   On your other ESPSomfy RTS device, do you see the frames being sent from the new one in its log?  If you do, then we know it is transmitting.  If not double check the wire you have connected to Pin 3 (GDO) of the transceiver and make sure the GPIO pin on the ESP32 is the one selected in the transceiver configuration.

Once we are sure of that and it is still not pairing we need to go back to the instructions for the motor.  From what I know of the Dexxo motors is that they have a couple of modes that they operate in.  They can be set up to pair a 3-button remote Up/Down/Stop or a one button toggle.   The manual that I have been able to find is a ridiculous array of hieroglyphs and cave paintings that you have to study.   For instance there is a 2 digit LED that has to be in a particular mode depending on the type of remote.

ThibautMeric, 2024-08-19:
That was a super fast answer! Thanks for that quick reply!

I like your technique to verify the device transmission. I would have loved to experiment around that, but unfortunately I don't have the older device with me and so I can't use it. 

I didn't realise that radio could be initalized and activated with transmission not working.

So, based on your reply, I did another wiring inspection focused on pin 3 (thanks for your reminder) and discovered that I made a mistake on the port mapping. (22 instead of 23 😅) and now everything is working fine.

Now I can proudly say that a Dexxo Pro 800 RTS can be remotely controlled with HA through ESPSomfy 😊

Once properly configured, the process of pairing was as straight forward as usual.
- In the app, set the frequency to 433.42Mhz
- Create a shade with the following configuration ( RTS / 56 Bits / Garage (3-button) ) 
- Press the programming button behind the main remote for 2 seconds. The lightbulb of motor will turn on and the 7 segment display of the motor will display F0
- Click pair shade
- Once done, the lightbulb of the motor will start blinking and you done 

Worked first time for me 😄 

Thanks again for your time

Thibaut

dokterdok, 2024-10-15:
Sharing my experience as it may help someone in a similar situation.

I made the mistake of copy/pasting a remote address from the radio logs into a new ESPSomfy remote without pairing it first, to give it a try. While it briefly worked, both ended up not working anymore, and I only realized after the fact that it was a serious mistake.

As I have a total of 10 somfy shutters with no dedicated breakers and hard to access motors, resetting the motor to factory settings wasn't a practical option, and I feared the worst.

I luckily still had a RTXCom remote set up in Home Assistant, which allowed me to send a program command to the motor from the HA developer actions section (with some ChatGPT trial and error to figure out the right command to send). 

As I couldn't re-pair my remote I had broken to the motor, I ended up swapping it with the remote of another room, and pair both again with their new motors.

Long story short: don't be like me and rush through the setup, carefully  read the docs first.
Food for thought:
If many people experience similar    issues, it might be worth adding a warning in the ESPSomy UI when attempting to edit a new remote address manually.