rstrouse/ESPSomfy-RTS #244: Room managment with home assistant

Pulpyyyy, 2024-01-17:
Hello Rstrouse,

I'm still using this great combo with home assistant and I'm still very satisfied.
I discovered that you can only select one zone/room for all rts devices.
[image]
I assume the room is inherited from the one assigned to the integration.
[image]
May it be changed, as obviously having all my covers in the same zone/room isn't the same as reality.

Regards,

rstrouse, 2024-01-17:
Yes you can change this by clicking on the entity then turning off Use Device Area.  After that you select the room that it is in.  I am adding new functionality with 2.3.1 that allows you to define rooms in ESPSomfy RTS but I am not sure I will push theses down to HA at this point.

rstrouse, 2024-01-22:
I assume you have gotten this squared away. v2.3.1 was released and it contains the ability to assign rooms in the UI.  I did not try to mirror this functionality into HA since I would never be able to keep them in sync.

closed 2024-01-22 by rstrouse

Pulpyyyy, 2024-01-24:
Thanks !
Just in case, room is size limited to 20 chars

rstrouse, 2024-01-25:
Yes it is limited to 20.  I am pushing against a hard memory limit on the ESP.  But realistically, most folks seem to replicate the room name in the name of the shade.  You don't need to do this as the room will appear above the name.