rstrouse/ESPSomfy-RTS #246: Loosing Ability to use ESPSomfyRTS device through HA

Coolbreeze68, 2024-01-18:
Hi,

Happy new year and thanks for this useful project !!

Actually Im facing an issue. At a certain amount of time Im loosing my Somfy buttons in HA, they are disabled.

[image]

The only way I found to enable them back is to untick and tick again "Publish discovery" and save.

[image]

Then my buttons are enabled again

[image]

I wonder if we could have a new param in the firmware to set to force send a Discovery message every x min ( like an heartbeat )

Or I might miss something I can do on my side.

Thx a lot 

Christian

rstrouse, 2024-01-18:
Update from Main using the Github update.  The MQTT configuration is persistent in the upcoming version.  Why MQTT-AD over the HACS component?

Coolbreeze68, 2024-01-19:
Tbh I didnt see that component exists :/. 
Looks like I cannot use HACS with HA Os. I'll give a try installing it manually.
Thx a lot anyway

rstrouse, 2024-01-19:
I use HACS with HAOS just fine.

rstrouse, 2024-01-21:
If you do some searching there are a number of tutorials out there to get HACS on HAOS.  It is pretty easy but it is a few steps.

https://www.makeuseof.com/home-assistant-install-hacs-custom-addons/

Coolbreeze68, 2024-01-22:
Yeah I've done my searching and I could install it. Indeed it is better that way thank you.

rstrouse, 2024-01-22:
HACS is a good option and the plugin supports many more features.

closed 2024-01-22 by rstrouse