rstrouse/ESPSomfy-RTS #247: Problem with groups

schumissimo63, 2024-01-19:
### Hardware

ESP32

### Firmware version

V2.3.0

### Application version

V2.3.0

### What happened? What did you expect to happen?

I have a problem with groups:
1) when I activate mqtt only the panes appear in HA, the groups do not appear.
2) if I created a group of 3 sections, no problem it works, I have 2 groups of 3 sections which go back into HA and which work correctly. On the other hand, if I created a group with the 6 shutters, I have one or two shutters which do not trigger.

I have the latest version, 2.3.0.

### How to reproduce it (setp by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-01-19:
You are correct.  Groups do not supported in the MQTT discovery.  This is because MQTT discovery doesn't have an entity domain concept for a group that I am aware.  Cover is not the same.

A group only sends out 1 frame + the number of repeats.  So if one or two shutters re are triggering, it means they did not hear the same frame that the others did.   Increase the number of repeats for your group.

schumissimo63, 2024-01-20:
Hi,

Thank you for your answer.
Ok for groups in MQTT, on the other hand I really have a problem with groups, I'll explain:
I did a clean installation with V2.3.0, I only restored my radio config, I reinstalled the shutters one by one, each one works correctly individually, but as soon as I integrate them into a group, it's the anarchy. I created the "balcony shutters" group, I integrated the "dining room shutter" into this same group, I tested and it was the living room shutter which is located right next to it which was triggered. I created the "living room shutter", I tested the group and it is the kitchen shutter which is right next to the living room shutter which is triggered. I integrate the "kitchen section", I test and two other sections open.
Before posting I redid the connection procedure described in this page https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#shademotor-setup but nothing to do the problem persists, it's is going crazy lol
I'm going to do all this again with V2.1.1, it worked when I installed it the first time.

schumissimo63, 2024-01-20:
After multiple attempts I found how to solve my problem, I changed the number that makes up the remote address and now it works great!!!! Many thanks for your superb work!!!!

rstrouse, 2024-01-20:
I was going to say that it sounds like you have a duplicate address somewhere.  Unfortunately, you cannot know what addresses are already stored in the motor since the motor cannot speak.  It only has a receiver and not a transmitter.   This happens when an address is paired to the motor and then not later unpaired.

rstrouse, 2024-01-20:
I am going to close this out for now.  Glad you got it all working.

closed 2024-01-20 by rstrouse