rstrouse/ESPSomfy-RTS #249: Memory overflow issue?

nl0pvm, 2024-01-24:
### Hardware

ESP32

### Firmware version

v2.3.1

### Application version

v2.3.1

### What happened? What did you expect to happen?

After adding the 29th shade things break :)

Symptoms:
- Shade 27 suddenly shows up as "Undefined", when editing that shade, the correct name is shown
- Shade 28 is "gone"
- Shade 29 is "gone" 

However the export shows the added shades correctly.

### How to reproduce it (setp by step)

```bash
1. add 28 shades
2. add the 29th shade and then things go sour ;)
```

### Logs

```bash
the .backup file:

 19, 66,   29,  7,  276, 29,  194,  0,   73,  111,   74,C6132C   
  1,Woonkamer           ,  1
  2,Kleine Slaapkamer   ,  3
  3,Kantoor             ,  4
  4,Overloop            ,  2
  5,Keuken              ,  0
  6,Slaapkamer          ,  5
  7,Zolder              ,  6
  1,true ,  4,   1023133,Kantoor Voorkant B  ,  0,  0, 56,     60000,     60000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,   18,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  1, 17,  0,  0,  0,  0,  3
  2,true ,  4,   1023134,Kantoor Voorkant O  ,  0,  0, 56,     60000,     60000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,   19,  0,    -1.00000,    -1.00000,   100.00000,     0.00000,false,false,  1, 18,  0,  0,  0,  0,  3
  3,true ,  4,   1023135,Keuken Klein Raam B ,  0,  0, 56,     45000,     45000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  1,  0,  0,  0,  0,  0,  5
  4,false,  4,   1023136,Keuken Klein Raam O ,  0,  0, 56,     45000,     45000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    3,  0,    -1.00000,    -1.00000,   100.00000,     0.00000,false,false,  1,  1,  0,  0,  0,  0,  5
  5,false,  4,   1023137,Keuken Rechts B     ,  0,  0, 56,     60000,     60000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    2,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  1,  2,  0,  0,  0,  0,  5
  6,false,  4,   1023138,Keuken Rechts O     ,  0,  0, 56,     60000,     60000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    2,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  1,  3,  0,  0,  0,  0,  5
  7,false,  4,   1023139,Keuken Links B      ,  0,  0, 56,     60000,     60000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    2,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  1,  4,  0,  0,  0,  0,  5
  8,false,  4,   1023140,Keuken Links O      ,  0,  0, 56,     60000,     60000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  1,  5,  0,  0,  0,  0,  5
  9,true ,  4,   1023141,Keuken Zijraam B    ,  0,  0, 56,     60000,     60000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    2,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  1,  6,  0,  0,  0,  0,  5
 10,false,  4,   1023142,Keuken Zijraam O    ,  0,  0, 56,     60000,     60000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  1,  7,  0,  0,  0,  0,  5
 11,false,  4,   1023143,Woonkamer Klein Raam,  0,  0, 56,     44000,     44000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  1,  8,  0,  0,  0,  0,  1
 12,true ,  4,   1023144,Schuifpui Rechts B  ,  0,  0, 56,     72000,     72000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  1,  9,  0,  0,  0,  0,  1
 13,true ,  4,   1023145,Schuifpui Rechts O  ,  0,  0, 56,     72000,     72000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  1, 10,  0,  0,  0,  0,  1
 14,true ,  4,   1023146,Schuifpui Links B   ,  0,  0, 56,     72000,     72000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  1, 11,  0,  0,  0,  0,  1
 15,true ,  4,   1023147,Schuifpui Links O   ,  0,  0, 56,     74000,     74000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 12,  0,  0,  0,  0,  1
 16,true ,  4,   1023148,Overloop B          ,  0,  0, 56,     49000,     49000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 13,  0,  0,  0,  0,  4
 17,true ,  4,   1023149,Overloop O          ,  0,  0, 56,     49000,     49000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 14,  0,  0,  0,  0,  4
 18,true ,  4,   1023150,Kleine Slaapkamer B ,  0,  0, 56,     60000,     60000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 15,  0,  0,  0,  0,  2
 19,false,  4,   1023151,Kleine Slaapkamer O ,  0,  0, 56,     60000,     60000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 16,  0,  0,  0,  0,  2
 20,true ,  4,   1023152,Kantoor Zijkant B   ,  0,  0, 56,     60000,     60000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    2,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 19,  0,  0,  0,  0,  3
 21,true ,  4,   1023153,Kantoor Zijkant O   ,  0,  0, 56,     60000,     60000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 20,  0,  0,  0,  0,  3
 22,true ,  4,   1023154,Slaapkamer Rechts B ,  0,  0, 56,     48000,     48000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 21,  0,  0,  0,  0,  6
 23,true ,  4,   1023155,Slaapkamer Rechts O ,  0,  0, 56,     48000,     48000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 22,  0,  0,  0,  0,  6
 24,true ,  4,   1023156,Slaapkamer Links B  ,  0,  0, 56,     48000,     48000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 23,  0,  0,  0,  0,  6
 25,true ,  4,   1023157,Slaapkamer Links O  ,  0,  0, 56,     48000,     48000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 24,  0,  0,  0,  0,  6
 26,true ,  4,   1023158,Zolder Voorkant B   ,  0,  0, 56,     61000,     61000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    1,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 25,  0,  0,  0,  0,  7
 27,true ,  4,   1023159,Zolder Voorkant O   ,  0,  0, 56,     61000,     61000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    5,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 26,  0,  0,  0,  0,  7
 28,true ,  4,   1023160,Zolder Achterkant B ,  0,  0, 56,     55000,     55000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    3,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,fal [... cut ...]
 29,true ,  4,   1023161,Zolder Achterkant O ,  0,  0, 56,     55000,     55000,      7000,  100,         0,         0,         0,         0,         0,         0,         0,    3,  0,    -1.00000,    -1.00000,     0.00000,     0.00000,false,false,  0, 28,  0,  0,  0,  0,  7
"v2.3.1","ESPSomfyRTS","pool.ntp.org","CET-1CEST,M3.5.0,M10.5.0/3",true 
  1,false,"192.168.0.65","192.168.0.1","255.255.255.0","192.168.0.1","8.8.8.8",  0,  0,  0,   0,  -1,  23,  18
true ,  0, 56, 18,  5, 23, 19, 13, 12,   433.420,    99.97,    47.60,  10
```

rstrouse, 2024-01-24:
That is a JSON serialization error.  Looks like I will be needed a bigger buffer because of the added rooms data.  I'll increase this buffer size and tell you when it is up there.  Thanks for the backup file as this will help me so I do not have to add them in one by one.

rstrouse, 2024-01-24:
If you do a github update from Main it should now be large enough to send all the shade, group, and room data to the client.

nl0pvm, 2024-01-25:
Sidenote: The upgrade to main did not go flawless. It does boot but the webinterface is unavailable. It does continu to respond to ping but no extra log-entries of (trying to) reload the webpage are created. 

I will now try to do a re-flash over USB.

[crashlog 20240125.txt](https://github.com/rstrouse/ESPSomfy-RTS/files/14050786/crashlog.20240125.txt)

nl0pvm, 2024-01-25:
> I will now try to do a re-flash over USB.

UPDATE: FIXED 
By reinstalling SomfyController.onboard.esp32.bin v2.3.1. Then it booted with an empty config. Then did the github upgrade to main. Then restored the backup with all checkmarks on. Then booted happily and fine

closed 2024-01-25 by rstrouse