rstrouse/ESPSomfy-RTS #250: ESPSomfy not booting

Baseje, 2024-01-26:
### Hardware

ESP32

### Firmware version

v2.3.0

### Application version

v2.3.0

### What happened? What did you expect to happen?

I've flashed an AZ-Delivery ESP32-WROOM-32 with ESPSomfy and it was working well.

After reconnecting the ESP to USB it looks like ESPSomfy wouldnt boot,
If i start ESPHome web, and open up the logs the boot process starts and the ESP is connected to wifi in a few seconds...

Someone an idea how to fix that the ESP always boot after USB power?
I've also the same ESP with ESPHome, and this boots directly after USB power...

### How to reproduce it (setp by step)

```bash
Put power in ESP, no boot.
```

rstrouse, 2024-01-26:
If it is connecting to wifi then there is some other issue.  Perhaps a corrupted filesystem but we can rule this out if you type
`http:<ipaddress of ESP>/downloadFirmware?ver=v2.3.1` in a browser and hit enter.  Let it sit for a couple of minutes and try again.

Baseje, 2024-01-26:
Thanks for your fast response!

After the ESP is connected to wifi, ESPSomfy is working well...
But without the action to start ESPHome Web, and go to the logs over COM, it wouldnt connect...

Output of the ESPHome Web logs;
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

Startup/Boot....
Mounting File System...
File system mounted successfully
Chip Model ESP32-
Preference IP Free Entries: 506
SECURITY   Type:0 Username:[] Password:[] Pin:[] Permissions:0
Connection Type: 1
NTP Settings 
pool.ntp.org TZ:CET-1CEST,M3.5.0,M10.5.0/3
WIFI Settings
 SSID: [***] PassPhrase: [***]

Launching web server...
Creating Web MicroServices...
WiFi Mode: 0
```

rstrouse, 2024-01-26:
I don't understand did you perform the step I outlined above?  The ESP will always boot when you apply power.  The log snippet above is the boot process.  The very next thing in line should be ESPSomfy RTS scanning all the networks to make sure the SSID exists.  Then it should start the socket server and connect to the AP.

Are you not seeing the available nearby networks in the log?

Calimerorulez, 2024-01-27:
I've got a similar issue.

My ESP32 stopped being accessible via the webinterface. Reboot did not help. I reflashed the onboarding firmware, hoping to resolve it, but booting 'hangs' after connecting to Wifi that I've configured on the landing page (after reboot):

```
Socket Server Started...
Max Heap: 110580
Free Heap: 173636
Min Heap: 169424
Connecting to AP
Set hostname to:ESPSomfyRTS
*******
Successfully Connected to WiFi!!!!192.168.3.169 (-46dbm)
MDNS Responder Started: serverId=A62C1C
Max Heap: 110580
Free Heap: 165944
Min Heap: 160436
App Version:2.3.1
Starting clean
Initializing RX Queue
Check github for updates...
Max Heap: 110580
Free Heap: 165892
Min Heap: 160436
```

rstrouse, 2024-01-27:
@Baseje please verify whether your board gets the Wifi networks or it gets stuck in the same spot as @Calimerorulez 

@Calimerorulez please install the firmware from the pre-release 2.3.2.  I removed the initial check for updates.  I am wondering if the github check is simply failing but the timeout is not working.  There are waaay too many topics on this timeout that seem to indicate that it will never return on timeout.  v2.3.2 removes this check for now and delays it for 1 day of uptime.  If you are running an esp32s2 I did not build for that this time.

Also, please make sure you are onboarding with the correct onboard file.  The chip series makes a difference.

Calimerorulez, 2024-01-28:
Hi @rstrouse,

Mine is booting OK now after updating to 2.3.2. Thanks again :) 
Running an ESP-Wroom-32 in this case, so I used the EPS32 onboaring file.

rstrouse, 2024-01-28:
@Calimerorulez please check on this once a day.  I am starting to wonder about the http client libraries on the ESP.  My connection to github is solid so I don't get failures when the firmware first checks to see if the internet is available then calls out to github to check the latest version.   I am wondering if the github site is not available whether it actually times out.

I am going to make a firmware that should fail when checking to see if it times out but I have seen reports online saying it can take half an hour regardless of the timeout settings.

Calimerorulez, 2024-01-28:
@rstrouse I can confirm it is offline again. With my other ESP build, based and build on an M5stack Atom, I don't have the issue. I don't know why.

rstrouse, 2024-01-29:
I am pretty sure I know why.  The check for the latest version is not timing out as it should.  This is an ESP32 idf issue.  I will have a workaround next week that doesn't check for a new version.  As much as I like that it notifies on a new version, having the device go down is not acceptable.  For me I don't see the issue but I suspect that has more to do with the git accounts that I have.

If I cannot figure out how to get the timeout to work reliably, I will remove the update check entirely.

rstrouse, 2024-01-30:
@Calimerorulez can you update to the pre-release v2.3.2 that I just published on the repo.  I found that the timeout for the secure client was 30 minutes so regardless of how short the timeout was for the http client it waited for half an hour before releasing it.  This is way past the time one would want to wait to see if the internet is available and way past the time the browser would wait around for its response.

So it just looked like it was not booting but it was booted.  It was just waiting around for a non-existent url.

Baseje, 2024-01-30:
After updating to v2.3.2 my ESP boots every time.

Only after an amount of time the webinterface isnt responding, but i've found issue #251 for this!

closed 2024-01-30 by Baseje

Calimerorulez, 2024-01-30:
@rstrouse 
Updated and running fine like @Baseje