rstrouse/ESPSomfy-RTS #251: ESPSomfy web page not loading properly

ngroves08, 2024-01-28:
### Hardware

ESP32

### Firmware version

v2.3.1

### Application version

v2.3.1

### What happened? What did you expect to happen?

When I navigate to the IP address of the ESP32, the ESPSomfy web page partially loads, before I get an (ESPSomfy) error message which says "Could not connect to server. Too many clients connected...." even though there is only one client connected.

The problem temporarily goes away after rebooting the ESP32, but returns after a period of time (maybe a few hours?)

This issues only appeared after upgrading to v2.3.0 from ~v1.7

Despite the issues with the web interface, I can still control the blinds via the Home Assistant interface.

### How to reproduce it (setp by step)

```bash
1. Boot up ESP32
2. Wait a few hours
3. Navigate to IP Address of ESP32
```

rstrouse, 2024-01-28:
I will look to see if I can create a condition where this exists.  If ESPSomfy RTS idle during this period or is it having activity like moving the shades or listening to a wind sensor?

Also I assume you are running an ESP32 not an ESP32-S3, C3, or S2, is that correct?

ngroves08, 2024-01-28:
Thanks

Re your questions:
- The device is idle most of the day. It controls roller blinds that open in the morning and close at night
- It's an ESP32-WROOM-32. I believe that's a plain ESP32. Let me know if that's not the case.

Also, the device is on a VLAN that has no access to the outside internet. Not sure if that makes a difference or not

rstrouse, 2024-01-28:
> Also, the device is on a VLAN that has no access to the outside internet. Not sure if that makes a difference or not

@ngroves08 the last statement may make a difference but it shouldn't.  I have read some issues with the ESP32 libraries that the HTTPClient libraries do not timeout like they are supposed to and ESPSomfy RTS is checking github to see if there is a new version available.   

What is supposed to happen is that it checks to see if the site is up then bails after 3 seconds if there is no return.  But I am guessing that this is not occurring based upon some reading.  The reports are conflicting in that some say it works as it is supposed to and others claim that it never times out.  I dill do some more investigation.

ngroves08, 2024-01-29:
Thanks @rstrouse 

Re internet access, I tried giving the ESP32 internet access. It went from not responding (when it had no internet access) to responding as expected (as soon as it got internet access).

Baseje, 2024-01-30:
> Thanks @rstrouse
> 
> Re internet access, I tried giving the ESP32 internet access. It went from not responding (when it had no internet access) to responding as expected (as soon as it got internet access).

I've exactly the same issue, after dropping internet the ESP comes not responsive after amount of time.

rstrouse, 2024-01-30:
Please ensure you have the absolute latest 2.3.2 pre-release and let me know the results.   I added some connection negotiation timeouts.  If this fails then the github check for update feature will die a painful death.  For me this works when removing my internet connection but there are a ton of ways to disable the outbound and inbound packets.  Some of which may confuse the ESP32 libraries.

ngroves08, 2024-01-31:
The new firmware seems to have fixed it. Updated this morning and web interface has been working properly ever since

Thanks!

Baseje, 2024-01-31:
I've just flashed to the latest version and disabled internet access.
I'll come back with my findings!

rstrouse, 2024-02-04:
I am pretty sure we got to the root of this issue.  I will be releasing a maintenance release v2.3.3 today to backfill a couple of other issues today.   Mostly cosmetic minor issues.

closed 2024-02-04 by rstrouse

TimelessNL, 2024-02-25:
> Also, the device is on a VLAN that has no access to the outside internet. Not sure if that makes a difference or not

I might think this is related, I was setting up my first `SomfyRTS` but after connecting it to my dedicated IoT network the device became unresponsive. it seems that the websocket ports are not correctly handled when being routed.
[image]
For testing I put my client device is on the `192.168.5.0/24` subnet while the `SomfyRTS` is on `192.168.255.0/24` with no firewall restrictions between the 2 subnets, when I try to access the `SomfyRTS` from within the same subnet everything works like expected, even without internet access on that subnet. But when routed it only loads the HTTP content.

For what it's worth, I'm using `v2.4.0`.