rstrouse/ESPSomfy-RTS #252: Wind sensor disappeared in Home Assistant after upgrade

pfak, 2024-01-28:
### Hardware

ESP32

### Firmware version

v2.3.1

### Application version

v2.3.1

### What happened? What did you expect to happen?

I upgraded to v2.3.1 and there is no Wind sensor entity exposed to HomeAssistant. I still see my wind sensor remote is paired.

[image]

[image]

### How to reproduce it (setp by step)

```bash
Upgraded from v1.7.2 to v2.3.1
```

rstrouse, 2024-01-28:
That is weird because I do not see the backup and reboot options in your display either.  Are there any errors in your logs and did you install the official release or the pre-release?

pfak, 2024-01-28:
The Home assistant component was on 2.3.0 when I updated the firmware and application to 2.3.1.

Order of events:

1.I was on HA 2023.5, updated to HA 2024.1
2. Noticed ESPsomfy HA component could not connect to ESPsomfy. 
3. Updated to 2.3.0 component 
4. Still could not connect/control via HA 

Few days go by... 

5. Updated ESPsomfy application and firmware from 1.7.3 to 2.3.1
6. Could connect/control via HA, no wind sensor entity 
7. Updated to 2.3.1 for the component, and I still do not see the sensor for wind.

There are no errors, and I installed only the official releases.

rstrouse, 2024-01-28:
First, go to home assistant an click reload on the device.  It will be under the 3 dots next to the configure button.  

If that doesn't fix it please upload a backup from your ESPSomfy RTS device.

rstrouse, 2024-01-30:
I will have a look.

rstrouse, 2024-01-30:
Do you only have a wind sensor and no sun sensor.

pfak, 2024-01-30:
I believe so. The installer was not too clear on it, but there's only one battery powered remote on the unit.

rstrouse, 2024-01-30:
For now edit the shade and check the sun sensor checkbox.   Then go to HA and reload the integration.

rstrouse, 2024-02-04:
@pfak update the plugin and the firmware to v2.3.3 and it will automatically add the wind sensor for awnings regardless of the sun sensor settings.

closed 2024-02-04 by rstrouse