rstrouse/ESPSomfy-RTS #254: Hardware ESP not getting to run. Tried 2 ESPs, 3 CC1101, still licking the frozen flagpole

fft1010, 2024-01-31:
Hi,
I am stuck, and I think it is the hardware pin on my ESP. I could get the Software to connect to my WLAN at home, I can read the Somfy remote controls with my Hardware (Up,Down, My, Pro), but if I want to send data, it says radio not initialized. I learned that this has something to do with the TX/RX. I really quadrouple checked for the last days, and now I am out of any more ideas what to do.

I have the CC1101 that looks excatly the same than this here on the docs, but my ESP is IMHO slightly different.

I have no real clue with hardware, but I assume that my pins on the board are different. See here the 
[image] Pin layout from the ESP board. The ESP board is from AZdelivery, a german based company.

The numbers are the pin numbers from the CC1101 tranciever.

Is there anyone that can enlighten me? Many thanks!
Juergen

rstrouse, 2024-01-31:
First lets verify which transceiver you have.  Take a pic or verify that it says E07-M1101D.  Typically, the radio not initialized message is either because the one or more of the 4 SPI pins are not correctly connected or the Enable Radio checkbox was not checked.  The radio will initialize if the RX and TX pins are incorrect.

fft1010, 2024-01-31:
Hi rstouse,
many thanks for the help. I can't believe it: It now works. I would have sworn, that I had the very same setup I have right now already tested.
Testsetup: Two identical systems than I run "Pair share" in one, and than I see the "Prog" at the other. Both see me somfy remote. So now I can dig further with my shades.
Again: Many thanks for everything. 

Juergen

fft1010, 2024-01-31:
Hi,
just tested the first shade. It worked immediately. Many Many thanks! 

Juergen

fft1010, 2024-01-31:
Anywhere I can sent a Pizza to or something similar?

rstrouse, 2024-02-01:
No need for Pizza.  Just glad you found some utility in it.

fft1010, 2024-02-01:
Hi,
ok, than I thought, I'll give you a good laugh and start the "worst first case" competition :-) See my "pre-alpha" to protect the system from the cats. I just put the Antenna to this place to only have one picture. The antenna has a 2m wire (round about) so I can set it everywhere I need it. Works very well.

[image]

Again: You really made my day/week I was totally happy yesterday to get everything running. There are still some little problems, which I now try to fix.

Thank you
Juergen

rstrouse, 2024-02-01:
That's fresh... and vacuum sealed.  More importantly it works.

DeIdstaaner, 2024-02-03:
Hi rstrouse,

sorry I might seem like a complete idiot, but I don't understand what happened.

Last summer I came across your project, bought an ESP32-WROOM 32 from AZ-Delivery, installed the firmware (worked like a charme!) configured everything, installed an update now and then, and it just worked. Brilliant!
Last week I realized after some months when I didn't change the system that there is a version 2.3.1. So I updated SomfyController.ino.esp32.bin to 2.3.1 from 2.2.1 which was the last I had installed.
The update went through, but it showed only the rotating dots. Reboot, no change.

After cursing I tried to start from scratch and used https://web.esphome.io/ website to install everything fresh. The flashing finishes with a success message, but I can't access the device. The Wifi ESPSomfy RTS never shows up.

Then I tried to install a ESPHome first by using the "Prepare for first use" button, and it worked. I could access the then available Wifi, I can enter the required IP and can access it via my home network. I can even do the OTA update by loading the SomfyController.ino.esp32.bin file. But then nothing works.

I also tried Arduino IDE 1.8, but probably didn't understand how to configure it correctly.

I now ran out of options solving this by myself.

What I don't get in the first place: Why did it work last summer without any (!) issues and doesn't do the trick now?

fft1010, 2024-02-03:
HI,
I am new, but to me it seems, there is still the old version on the ESP. I personally used the esptool to install it. My main problem there was do get the ESP into download more. 
I only could get it into boot more by pressing the boot button on the ESP, THAN pressing boot button and release it, still keeping boot button pressed. 
Than start the esptool like described here:
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Installing-the-Firmware#using-esptoolpy
including flash_id and erase_flash, than flash the newest binary from the repository (2.3.2)

For all of the three commands above, I had to put the ESP into download mode.

Than you will IMHO have a clean new version, and can continue here:
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software

thanks
Juergen

DeIdstaaner, 2024-02-04:
Progress report.
I finally found out how to use Arduino IDE correctly (or at least better) and managed to install my old firmware again, restore the settings and have it at least basically working. During that process I found an error in the code which stopped Arduino compiling. I'm not really into this, but it where 2 problems with the call of elechouse_C1101.setGDO. One transmitted only one parameter, while it expected two, and on call was misspelled as ...setGDO0. I solved that by reasoning what could be righht and it worked. But I must admit that I didn't fully analyze what this function is used for.

Now I don't have the courage to try updating it again...

rstrouse, 2024-02-04:
@DeIdstaaner  If you are using the Arduini IDE I suspect you have an old version of the CC1101 library installed.  The signatures for these methods are as follows.   setGDO is called to set both GDO0 and GDO2 whereas setGDO0 only sets a single GDO0 pin.   ESPSomfyRTS will multiplex on that pin.

Just curious why you went the route of using Arduino.  Onboarding from the ESPHome website is very simple if you use the proper zip file containing the onboard firmware for your specific chip series (as indicated in the instructions).  
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Installing-the-Firmware

However, if you attempt to use ESPHome and flash only the firmware .bin file you will not have all the necessary partition data to run the software.

```
  void setGDO(byte gdo0, byte gdo2);
  void setGDO0(byte gdo0);
```

fft1010, 2024-02-05:
@DeIdstaaner ,
I am a real rooky with ESP-Somfy-RTA and HA. But _if_ you have the hardware running, and know that it works on the "physical" layer, and if you now have a running system that runs "half way", I would
a.) make a backup of the settings
b1.) Buy a new ESP and flash it from scratch (the non courage version), install it with WLAN (_really_ easy!! The software works like a charm. I did it one week before, and it is really absolutely straight forward.) Install the backup from above, exchange the old ESP with the new one. Used the phone to hook it to the ESP, configured there the local WLAN, disconnected and ESPSomfy showed up. (I think a minute or so max)
OR
b2.) flash the old over and install the backup. Configuration to get the ESP to hook up into the lokal WLAN is a 2 minutes job.

For flashing I personally would use shell and esptool.py, as there you IMHO see more. I could not get the browser install  work, and it only said: DOES NOT WORK. (something similar). When on the shell, I could directly use a termin (putty on windows) to see if the ESP is "there", and is listening for download. I also saw the three commands that esptool was running and could step thru and see, if every command worked.

My major problem was to get the ESP to get in download (program) mode (also have an AZ delivery one.) I could only make it to get in download mode by 

press boot button on ESP (keep pressing) -> press and release EN button on EN (still holding boot button). 
Start esptool and release boot button when esptool tries to stablish a connection.

than to all three epstool command I was referring to above. Than you have a brandnew ESPSomfy-RTS, you need to put again into the WLAN, ... But again: This is really easy.

thanks
Juergen

> Progress report. I finally found out how to use Arduino IDE correctly (or at least better) and managed to install my old firmware again, restore the settings and have it at least basically working. [...]

luckyzor, 2024-02-10:
Hello, I'm having the same issue with exactly the same board and cc1101.
What did you do to put it working??
I've already checked the connections twice and still no radio connected :/
don't know what to do

rstrouse, 2024-02-10:
@luckyzor In 100% of the cases where the radio would not initialize so far it has been a wiring issue or the user forgot to check the enable radio checkbox.   The radio will not initialize if the SCLK, CSN, MOSI, or MISO pins are not connected to the correct pin on the transceiver.  I assume you have an E07-M1101D transceiver.  All of these so far have been blue in color.

There are 8 pins and it is easy to connect the wrong pin on the transceiver to the wrong pin on the ESP32.   You need to then match these up to the configuration on the transceiver page.  Take some pics of your board and your transceiver and I will help you find your issue.

luckyzor, 2024-02-10:
Well I changed the esp32 and it worked.
When investigating the esp32 not working, there was an issue with the soldering on one of the pins.... Well now I have 2 esp32 with espsomfyrts xD
Hope I don't need 2, I really don't live in a bunker 😅

Thanks for the help and sorry, it was a hw issue at my side.

My somfy motors will be installed in 4 days, hope this will work, they will install 5 rts somfy solar blinds and 4 rts hybrid somfy blinds (Its a mix of rts and physical button, they call it hybrid)

rstrouse, 2024-02-10:
@DeIdstaaner did you get up and running?  I am afraid that we have different issues going on in the same thread so I missed what you were doing.   First, you did not download the `SomfyController.onboard.esp32.bin.zip` file.  You cannot initially flash with the files that end in `ino.espxx.bin`.   Those do not contain all the items to make the software work the first time.  They are only for updates.

Next you installed ESPHome stub software.  The instructions say to ignore the prepare for first use button and click on install.  Then select the extracted bin file that matches your chip series using the install button.  Please review the wiki and let me know if you have any questions.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Installing-the-Firmware

DeIdstaaner, 2024-02-18:
Success!
That was my mistake. Since the ...ino.espxx.bin files inside and outside of the zip have the same name I falsely assumed that they where the same files. But as rtsrouse wrote these are different!
After extracting and installing the ...bin file it worked flawlessly with the esphome.io website, I could restore the previous update and update it to the newest version.
As it is very often the case: the problem was between my ears....
Thanks

rstrouse, 2024-02-19:
Awesome!  I am going to close this one out.

closed 2024-02-19 by rstrouse

Amitcooo, 2024-08-10:
@rstrouse Hey, 
I'm trying to extend my range for a homemade rc controller that I'm working on to my homemade esp32 rc plane, I used to communicate between them with the esp-now protocol but I need it to revive and transmit for longer distances so I'm working on changing the communication system to CC1101 v2 chip. I bought it from AliExpress but every time when I'm try to make it work I fails and can't find out what's the problem. I'm straggling to fine information about that in the internet and I don't even know if it because of the wiring of the esp or even the code itself. I'm wondering if the chip just isn't working because at the time that I received it I connect it multiple times to the Arduino nano 5v logic spi and I don't know where even I need to start with to troubleshoot this problem and make it work. I'm so close to put it back in the box and forget about it because I'm so confused and have no motivation to continue. Can you please help me find the problem and troubleshoot it?? Please even if you will just precise me a wiring guide and basic instructions to make that model work for the first time it will be amazing!!!

rstrouse, 2024-08-10:
@Amitcooo I believe you are trying to post to the wrong repository.   This software does not control rc planes.  It is for window coverings.

Amitcooo, 2024-08-10:
> @Amitcooo I believe you are trying to post to the wrong repository.   This software does not control rc planes.  It is for window coverings.

Yeah I know the rc plane story is just a background about my project but I'm new hear at GitHub so sorry if I did something wrong for hear cause I don't know the rules...

rstrouse, 2024-08-10:
There are no rules but I’m afraid I cannot be much help to you with a protocol for RC planes.  Although, I still have two third scale RC biplanes.  These use a PCM 2.4 radio.

Amitcooo, 2024-08-11:
> There are no rules but I’m afraid I cannot be much help to you with a protocol for RC planes.  Although, I still have two third scale RC biplanes.  These use a PCM 2.4 radio.

I have a problem with my esp32 after connecting it to the model... I tried all the flash factory rest methods and I don't know what to do now... can you help me?

rstrouse, 2024-08-11:
Unfortunately, I cannot help you.  This repository is for window coverings not model airplanes or robotics.  Below is a link to the readme for the repository.

https://github.com/rstrouse/ESPSomfy-RTS?tab=readme-ov-file#espsomfy-rts-