rstrouse/ESPSomfy-RTS #255: Adding setting for automatic GitHub update check

passibe15, 2024-01-31:
### Describe you new feature you'd like

Following #251, the newest 2.3.2 pre-release seems to have fixed the issues relating to no internet connectivity for me as well (smoothly running for about 18 hours now).
However, from my firewall logs I can see that the ESP device is trying to connect to GitHub (for me its IP addresses 140.82.121.3 and 140.82.121.4) every four hours.

So as a feature request:
Would it be possible to simply add a checkbox e.g. on the firmware page saying "Automatically check for updates" (or something along those lines) to disable this?
With the checkbox enabled the timer would not run and e.g. requests to github would only be made when clicking on "GITHUB UPDATE", while then displaying an error message/retry button if GitHub is not reachable.

Thanks!

rstrouse, 2024-01-31:
Yes that feature is already in the works.  I did set the timing for 4 hours so that is correct but the final will only check once a day.

rstrouse, 2024-01-31:
The latest pre-release v2.3.2 has this feature.

passibe15, 2024-01-31:
Wow, that was quick. I just installed it and everything seems to work so far.
Thanks a lot!

closed 2024-02-04 by passibe15