rstrouse/ESPSomfy-RTS #256: Telis 4 Remote - Shade Status with Group Remote

ecastete, 2024-02-01:
I searched the issues and didn’t find any notes about this software configuration scenario.

I have a Telis 4 remote with the fifth position being group mode.  When I sent commands from th remote to the group of sheds the status change was not shown in the ESPSomfy-RTS UI.  

So, I went to each shade and added another remote by pressing a burr on on the Telis 4 while it is in the group mode.  So now, each shade shows two remotes in ESPSomfy-RTS, one for the individual shad/remote button nd one for the group shade/remote button.  As a result, the shade status in the UI is correct whether I use the individual or group mode on the Telis remote.

[image]

[image]

In the image above you can see a common remote number for each shade. This is the group mode from the Telis remote.

Hope this is helpful and not a duplicate.

rstrouse, 2024-02-01:
Yep that is the correct way to link the grouped channel from your remote.  When I get some time I have some documentation updates (additions) to do.   I'll do better to indicate how to set up a grouped channel.

rstrouse, 2024-02-07:
I added some additional instructions to the Wiki.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#linking-remotes

closed 2024-02-07 by rstrouse