rstrouse/ESPSomfy-RTS #257: 12db 433Mhz Antenna

ecastete, 2024-02-01:
In case someone is trying to cover a larger area this 12db antenna may be useful.

https://www.amazon.com/dp/B086TW5SYX?psc=1&ref=ppx_yo2ov_dt_b_product_details

rstrouse, 2024-02-01:
Is this antenna working for you?  Even according to their pictures on Amazon this antenna looks to be tuned between 470 to 480Mhz.  Oddly they advertise it as a replacement for a router antenna as well.  Those two frequencies are not even close to each other.  I bought a lot of antennas including some that claimed they were 433Mhz antennas but were really surplus wifi or GSM antennas.

It could just be that some of their marketing materials are wrong.  Then again their router picture is actually a switch that they pasted the antenna in front of.  So the seller may not have a good idea of what they are selling and the only answer is whether it is working for you.

ecastete, 2024-02-01:
It is working but let me do some additional testing again comparing it to the 3db antenna in your Simple ESPSomfy RTS setup.  I looked at their graph again on Amazon and it is measuring SWR across frequency spectrum.  SWR at 433MHz is 5 but it is down to 1.9 at 470MHz.  Interestingly, if you look to the far right of the graph the SWR is even lower around 900 MHz. I’m beginning to think it is actually designed for 950 MHz and SWR is 1.9 at 470MHz because that wavelength is 1/2 of 950MHz.

This is dragging up some of my RF knowledge that I haven’t used seriously in the last 30 years. 😋

More to come…

ecastete, 2024-02-01:
One other thing to note, and they have a good pic with that antenna. 

[image]

People need to be aware when they purchase antennas that they don’t buy a RP-SMA connector.  Screwing a Reverse Pin SMA male antenna onto the CC1101 will NOT work.  It wouldn’t surprise me if some of the signal issues people have with a new build are because they have a RP-SMA antenna.  This blog post. provides a good explanation.  

https://blog.linitx.com/what-are-sma-rp-sma-connectors-and-whats-the-difference/

rstrouse, 2024-02-07:
I am going to close this one for now.  If you find out more information feel free to re-open it or provide your findings on a new ticket.

closed 2024-02-07 by rstrouse