rstrouse/ESPSomfy-RTS #258: Webinterface unavailable after upgrade from 2.3.1 (Main) to 2.3.2

nl0pvm, 2024-02-03:
### Hardware

ESP32

### Firmware version

v2.3.2

### Application version

v2.3.2

### What happened? What did you expect to happen?

After the update from 2.3.1 (i was running the main because of [issue 249](https://github.com/rstrouse/ESPSomfy-RTS/issues/249) the webinterface and HA-interface is unavailable. After the first reboot the HA-interface is available.

The following config was used pre-backup
[ESPSomfyRTS 2024-01-30T00_11_59.backup.txt](https://github.com/rstrouse/ESPSomfy-RTS/files/14150904/ESPSomfyRTS.2024-01-30T00_11_59.backup.txt)

### How to reproduce it (setp by step)

```bash
1. run 2.3.1. (MAIN) with the config provided
2. press the update from github button 
3. upgrade to 2.3.2 
and wait :P
```

### Logs

```bash
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################

					POWERON log

########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
########################################################################################################################################################################################
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[... 135 lines cut ...]
Free Heap: 172112
Min Heap: 170036
App Version:2.3.2
                 shades.cfg exists so we are using that
Reading header at 0
                   version:19 len:66 roomSize:29 roomRecs:7 shadeSize:276 shadeRecs:29 groupSize:194 groupRecs: 0 pos:66
                                                                                                                        Initializing RX Queue
Timing Net: 5010ms
                  Check github for updates...
Max Heap: 110580
Free Heap: 172104
Min Heap: 164472
Internet is Available: 1719ms
                             [HTTPS] GET... code: 200
                                                     [HTTPS] GET... code: 200 - -1
                                                                                  Max Heap: 110580
Free Heap: 168816
Min Heap: 117044
Loading file /index.html
Timing WebServer: 8825ms
                        SSDP: urn:schemas-rstrouse-org:device:ESPSomfyRTS:1 - true
                                                                                  Connected to SSDP...
Timing Net: 2696ms
                  Loading file /index.js
```

rstrouse, 2024-02-03:
Enter the following into a browser `http://<ip of espsomfyrts>/downloadFirmware?ver=v2.3.2`.  It will redownload the firmware and application.  This will fix the application files.

nl0pvm, 2024-02-11:
Last update to 2.3.3. worked flawless so.. should we concider this bug closed?

closed 2024-02-12 by rstrouse