rstrouse/ESPSomfy-RTS #259: One shade behaves different when called from settings->somfy->groups->edit group than the group from home

fft1010, 2024-02-03:
Hi,
I have installed today all newest versions. HA and ESPSomfy-RTS and ESPSomfy-HA. Deleted them before, installed new, according to my knowledge, it is "fresh". Here my problem I have:
I have 8 shades in the living room. I want to control with HA and ESPSomfy. I first tried to have a switch in HA that would use "Service" to call the Services for each 8 shades. It worked for 7. One shade (L3) was not reacting to "up", but to down and "My". OK, I than thought I setup a group in EPSSomfy directly, teach all shades, and call than this group from HA.
Now L1 (another shade) has the following behaviour:
a.) When I have all 8 in a group L1 is reacting to Up/my/down like all other perfectly it I am calling it in the "adding group mode" from settings->somfy->groups->edit group. It works in 99% or all case. (ok, really did not sample 100, but I would say it works "always").
b.) If I call the same Up/my/down from "home" from the L1 only reacts to Up and down reliably, but ignores "my" in I would say at least 80% of the case. _Sometimes_ it stops. 

I than deleted 3 shades (always kept L1)

c.) If I have 6 shades in the group (with L1) it works always also from "home"
d.) If I have 7 shades in the group (with L1) L1 did not react to anything any more from "home", but works perfect from settings->somfy->groups->edit group

Any idea why this is happening? It really puzzles my, why calling this group from "Home" of ESPSomfy is different than calling it from "configuration mode" at settings->somfy->groups->edit group

thank you
Juergen

fft1010, 2024-02-03:
Hi,
I nailed id down: After clearing the cache in ESPsomfy, everything works as expected.
I donno if it would be a good idea to clear the cache when something in settings is saved.

thank you
Juergen

rstrouse, 2024-02-04:
@fft1010 I have added some code to make sure the endpoint always reloads with current data from the ESP for the group and shade configurations.

closed 2024-02-04 by rstrouse