rstrouse/ESPSomfy-RTS #261: Repeat press of key when linked physical remote is recognized by a certain shade.

fft1010, 2024-02-04:
### Describe you new feature you'd like

Hi,
I have a feature request, that would help hopefully not only me, but also others. 
My setup: ESPSomfy (working) and still using physical somfy remotes (needed by the management, AKA my wife). I linked all remotes to the EPSSomfy and so shades in ESPSomfy are notified when the external remote is used.
Works, in the way: ESPSomfy gets every press from any remote. Does not work, in the way, that not every shade recognizes every remote key press any time, as one shade is often behind some concrete and iron, so this shade is not getting every remote key press (was always that way.)
So here the idea: 
Add "repeat remote press" number field at settings->somfy->shades->individual shade config
If there is a number added which is bigger than 0 (say N) any remote key press is repeated by ESPSomfy for this shade "N" seconds after ESPSsomfy recognizes the linked remote key press.
Reason: If the repeated key press comes to fast, I think that the "my" key behaves different.

I hope you like the idea.

many thanks for considering

thank you
Juergen

rstrouse, 2024-02-04:
Lets start backwards with the My key.  What you are likely experiencing is how the motor sets and clears the my position in the motor.   While it may appear to be based on time, it actually seems to be  based on the number of repeats received with the same address and rolling code.  If it gets too many repeats then it thinks the user wants to set or clear the current my position.

If you are looking to repeat commands that are originating from ESPSomfy RTS then the Repeat Commands dropdown will currently allow you to do this.   The benefit with this is that that a sleeping motor sometimes is lazy to get out of bed and simply repeating the command a few times keeps it from hitting the snooze button and going back to sleep.  This function already exists for shades and groups.

However, if the command is originating from the remote then this repeat occurs by holding down on the button.  The unfortunate part of the Somfy remotes are the abysmal range.   I suppose for most applications the expectation is that you will be in line of sight for the motor.  The placement of the ESPSomfy RTS device is more flexible so you can often move the transceiver to a location where it is more likely to actually receive a signal from a whispering remote.

Repeating commands from the remote would likely create other issues though.  This is because of the broadcast nature of the frames.  There is no real way to tell if another frame is coming after the current frame so if ESPSomfy RTS switched to transmit then this incoming frame would be clobbered and both the repeated frame and the original frame would be lost.  I'll keep this in mind

Somfy does sell an RTS repeater but I am unsure of its function.  I do suspect that it probably waits for silence before repeating there is probably some noticeable delay on the magnitude of .75 timing of a frame.  I can't guarantee anything as I am nearing the code limits on the ESP32.

fft1010, 2024-02-05:
Hi @rstrouse ,
I am referring to the following: Somfy original remote is pressed, and L1 shade is not listening to the Remote whisper:
[image]
Above is the layout of the room L1-4 and R1-4 are my shades, S-R is the original Sonfy Remote fixed to the wall (and it the place, where you enter the room, so the "natural" place in the morning to raise all shades). It it supposed to always put all shades up/down/stop (it has three buttons). ESPS-RTS is where I placed EPSsomfy-RTS. As you see: L1 can barely listen to S-R. I would say in 90% of the cases it does not react. We did not care the last 15 years, as to me the fix would be obvious, but not really worth it. We also have two more remotes, where we can steer L1-4 as a group and each individual and same for R1-4. So if we really do want to close or open this L1 shade, we do it there.
_BUT_ now as I see that ESPSomfy-RTS always, I thought of that idea to repeat the somfy remote press with ESPSomfy-RTS.
Small side problem: As the HA "remote" is too smart, it looks like this
[image]
in the morning, assuming that L1 is open. In fact it is closed, (pressing down/stop/up) does the trick, but management says: I though it will be better, it is only different :-)
Anyway: I have no clue, if my idea has other side effects, or is for other reasons not worth implementing, I just wanted to "campaign"(?) for my idea and give the full story behind.
Many thanks. The other workaround could be to ask HA, if 7 shades are up, and than open L1, ... or simply keep it the way as it is.
Again: No hard feelings, many thanks for ESP-Somfy-RTS.

thanks
Juergen

011V32, 2024-02-05:
Ideally, there is a function in ESPSomfy-RTS that forwards the commands and you also remove the remote control from the shades. This means that the shade only receives the command once. But support for such a function would be necessary in ESPSomfy-RTS...

rstrouse, 2024-02-05:
Actually, I am intrigued.  I have my own little 10 device echo chamber and a bit of code I wrote to relay/repeat/forward any incoming frames.  I will run some tests to see if I can do this without it turning into a shouting match between devices.  That is the fundamental issue as I suspect it might be beneficial to have more than one device that can mesh.

fft1010, 2024-02-06:
Hi @rstrouse ,
I do not understand, what issues you are expecting within the frames and echoing the radio signals from Somfy (if I understand the text above correctly). As I am totally dumb in this radio frequency area, my life is easy :-):

I press a button on my Somfy remote, ESPSomfy-RTS listens (as it already does, because the "shade up grafics" work, and remote is linked), and if EPSSomfy recognises that one shade is "usually not getting it", it repeats the Information it got from the remote let's say 2 seconds after it grabbed it (when putting up the grafics).

So: Where am I wrong? Or what can I learn?

Many thanks for beeing "intrigued" in this case (learned a new word to my school english) :-)

thank you
Juergen

rstrouse, 2024-02-07:
The issues with repeating a frame is simply that ESPSomfy RTS can only listen or speak at any given time.  When it is speaking it cannot listen and when it does decide to speak it stops listening.  So the feature needs to wait for a gap in the incoming packets to send out the frame.   Theoretically speaking the frames come in at 170ms including the silence that needs to exist between each frame.  

So what this new feature does is... it builds a queue of identified frames that it will send within a silent period.  When a frame is heard that needs repeating it will place that in the queue and wait for an opportunity to shout it out.  Still this is not perfect as remotes tend to send more than one frame and each of them need to be considered.  So if there is currently a frame being processed it will hold off on sending it by waiting 100ms after the incoming frame has been received.  If any frames are in the air when ESPSomfy RTS is talking then it will not hear those.

To identify a remote that should be repeated go to the Somfy section in the configuration and choose the Repeater tab.  From there click the Scan Remote button.   Then press any button on the remote you want to repeat and it will start repeating every command that it hears from that remote each time it hears it.  I set it up to only repeat up to 7 remote addresses but I think if you need more than that then the radio traffic might get to be too much.

fft1010, 2024-02-08:
Hi @rstrouse ,
after the hint with the new relase I got it :-) I did not think you where sooooo fast.
Anway: I installed the new version, paired the remote, and rebootet. To make it short: All old functionality I tested worked, the L1 still does not move.
I double checked, that L1 always moves with ESPSomfy, and it does. I also tested the remote up more than once to avoid side effect or sleeping L1. No effect.
My "feeling" is, that maybe the L1 even gets the signal from the ESPSomfy too fast after it is recognized. Can you make a "wait time" to see if this is the case?
Other idea: I have here soon (hopefully at the weekend) another Hardware for ESPSomfy-RTS. I could setup those and send you logfiles from those. If the "general logfile" from a regular ESPSomfy is fine I am all set, and would simply try it, maybe it gives some insights, if you have a special firmware in your eco chamber, I would need this. Version says 2.4.0

thank you 
Juergen

011V32, 2024-02-08:
Did you try if it works when the shade is unpaired from the remote?
(At least one other remote should be paired, but if I understood you correct, the ESPSomfy-RTS is paired so no problem to unpair the original remote from L1)

fft1010, 2024-02-08:
HI @011V32 ,
To make sure: Really unpair with the remote? I do not really now how this is to be made.

I can try to see what is happening, but as it is a group, that is called, from this remote, and L1 is the only one that is not reacting, I first have to find out, how later on, I get this L1 again to this group. If not, Management would be unhappy :-)

I'll check , if I can find the docs for the remotes. (Was paired 15 years ago from a service company.)

thanks
Juergen

011V32, 2024-02-08:
Hi @fft1010 
I have to admit that @rstrouse still needs to help here. Because if it is a direct forwarding, then decoupling is actually bad.
But if it's recording a command and then executing a new command, then that's how I would do it. Because then you can ensure that the signal is only recorded once for the shade.

To your question above:
Normally you use a remote control that only operates one shade. (You wrote that you have two additional remote controls where you can also control subgroups and individual shades) Make sure that only one shade is selected for multi-channel remote controls. Press the reset button until the shade makes a short up and down movement. (The motor is then in teach-in mode.) Then you press the reset button of the remote control that you want to tech in or decouple. (In your case then the reset button on the S-R had to be pressed)
Once the S-R has been removed, this shade can still be operated via the additional remote control or ESPSomfy-RTS.
You have to have done this before to teach in the ESPSomfy-RTS...

fft1010, 2024-02-08:
Hi @rstrouse and @011V32 ,
maybe something totally different is wrong. If I press "all up" on the physical remote, than all shades go up, except L1. If I than wait and press up on the ESPSomfy for L1 only, than it works.
So IMHO the "UP" for L1 is never fired from ESPSomfy, or too early(?).
So to verify this, a blunt "waiting time" of lets saya 1 second until RTSsomfy repeats a command would be fine, IMHO.
In our test scenario there is never more than one remote pressed at the same time.

thanks
Juergen

rstrouse, 2024-02-08:
It is a forwarding relay so it is not recreating a new command.  But it is also not direct since any crosstalk on the frequency at the same time will cancel each other out.   Sending a new command is also not a great idea given the repeat nature of the protocol.   

The wait states should be as short as possible but we an play around a bit with that.  I am pretty sure that the re-broadcast is happening as logs for every device on my desk received the frames when it relayed.   My desk is full of devices loaded with standard ESPSomfy RTS firmware and each device hears frames from the other devices.  

Just to be sure, you should have simply scanned your remote into the screen below without any other changes.  No pairing or unpairing required.

[image]

fft1010, 2024-02-08:
Hi @rstrouse ,
In the above I scanned the physical remote, that is "hidden" and causing the problem_with_ L1.
To be honest, I do mot understand why this is "OK", I see it as a "L1" problem, so I see a pair from remote/L1 which should repeat.

I am afk for about 6 hours, sorry, real life is asking for presence.

thanks
Juergen

rstrouse, 2024-02-08:
Part of the issue here are simple vectors.  These radio signals can only travel in a straight line.  While 433mHz will penetrate concrete, this often has rebar in it and if it is stucco some sort of wire mesh.   Sadly, signals travelling through that will suffer the fate of the Faraday cage.  

Keep in mind every motor in your home is listening for every frame.  In the Somfy RTS world you are either a receiver or a transmitter.  So every motor is a receiver and every remote is a transmitter.  ESPSomfy RTS is a transceiver, meaning it can both receive and transmit but not at the same time.  So while it is listening it needs to keep its mouth shut and when it is speaking it cannot hear anything.   To be fair that does not matter since two transmitters sending out data at the same time will destroy each other's signal.

So given that scenario when you press S-R it sends out a signal in all directions, resonating from the embedded antenna on the remote.  Depending on the location of the receiver antenna on L-1 it appears that the signal has to traverse through nearly 2 meters thick of obstruction if you draw a straight line to the antenna.   Bear in mind R1-R4, ESPSomfy RTS, and likely L2 and L3 probably hear that signal just fine.  They simply ignore it because it was not addressed to them.

Since ESPSomfy RTS can reliably hear that remote channel, it simply checks its address list to see if it is marked as a repeater whenever a frame comes in.  If the address is on the list, it queues it up so that when everybody is done talking it fires up its transmitter and relays that same frame to everyone around it.   If they are not interested in that address then they will simply ignore it but L-1 should finally hear that signal.  Based on how busy the traffic is on the radio waves this can happen as early as 140ms from the time the frame was first processed.

Given the diagram below, the purple arrow represents the original frame and the orange one represents the repeated frame.

[image]

rstrouse, 2024-02-08:
Also, given my description above, never, ever, assign the same repeat address to two different ESPSomfy devices at the same time.  They will repeat themselves into perpetuity.  Each one will echo the frame to the other in a game faster than professional ping pong players.  The RTS protocol does not contain the source of a transmission only the destination so no receiver can know who sent the frame and each one will repeat that same frame over and over again.

fft1010, 2024-02-09:
Hi,
ok. It seems everything works. I setup another ESPSomfy RTS and that clearly showed that the first ESPSomfy resents the same information some ms later.
I than ripped of the "Wall remote" and found out, that L1 seems not to be in the "all up" group (Standing in front of L1, all other 7 did their job, but L1 was doing nothing, Original ESPSomfy repeated). Maybe during pairing there was a problem, and nobody checked it, and we thought for 15 years (sic!), that L1 is "behind the chines wall".

I have another remote where all 8 shared go up, this works, 4 shades go up: Works as well. Always the remote information is shown on my "listening" EPSSomfy and duplicated.
I now have to find out, how to put this L1 into the already existing group. Sorry for all the hassle. I would have _NEVER_ expected this.

thank you
Juergen

011V32, 2024-02-09:
Hi @fft1010 
I thought you said it sometimes works with the remote control?
> I would say in 90% of the cases it does not react.

@rstrouse
Except that this problem no longer exists, but wouldn't it make more sense to make one command into another command?
So that you can listen to something and then turn it into something else?
Then the issue of a loop with several devices would be off the table and the situation could always be clearly depicted. Because if a device only responds sometimes and you then send the same command again, this will cause problems with the My/Stop button. Or am I wrong?
But if you don't accept the actual command at the receiver, but only the new one, then you have a clear picture of the state again...

fft1010, 2024-02-09:
@011V32 @rstrouse ,
what puzzled me and this is why I always was asking for the _combination_ of the repeat:
I was hoping for: ESPSomfy hears "Up/down/my press specific remote", and know in that case: do "Up/down/my" for specific shade(s). Like individually repeat the command from this remote for individual shades, as the I as a human know that they have usually problems.
With my "little debug second EPSSomfyRTS" I saw that it is a "pure" repeater. Which is not what I expected. (Not judging in any way).
I donno if the above suggested feature does has side effect, or is hard to implement or whatever. I _think_ my feature gives more flexibility, but it might also give more chances to shoot yourself in your feet. _PLUS_ I value the time of others very much, and have absolutely no idea, what it would cost in time speaking. I just can say: I am willing to debug (that is what I might get right), I can't help here with programming.

anyway again and again
many many thanks for this Hard/Software and your endless support @rstrouse 

thanks
Juergen

rstrouse, 2024-02-09:
> Then the issue of a loop with several devices would be off the table and the situation could always be clearly depicted. Because if a device only responds sometimes and you then send the same command again, this will cause problems with the My/Stop button. Or am I wrong?

That is exactly why you cannot simply send a new frame series.  It has to be a repeat of an existing frame.  Otherwise it will get translated into an new command.  This problem is would rear its head for every button depending on the motor series and the tilt methods used.  Because it repeats the command this should not be a problem for the My/Stop.  Once the motor picks up the requisite number of repeats it has a cooling off period as it processes the command and the second set of repeats will be dropped.   I only hang on to 5 so it never gets into a dual long press.

> I donno if the above suggested feature does has side effect, or is hard to implement or whatever. I think my feature gives more flexibility, but it might also give more chances to shoot yourself in [...]

The RTS protocol has a facility to repeat commands.  In fact if you hold down a button on the remote that is exactly what the remote does.  And that is exactly what this feature does.   A new frame series should never be generated out of whole cloth when the expectation is to relay the original signal.   Generating a new frame series would have two for sure side effects.   The rolling codes for the originator would go out of sync and the resultant command would be misinterpreted by the motor as another command it if heard it the first time.

The My key (and the prog key) are an interesting thing.  It has nothing to do with timing rather the motor counts the number of repeats that it hears.  Because the frames are sent out in a specific timing length it can get distinguish between a long and short press.   For instance if a new frame series was generated from the remote, holding down the My button would both set and unset the My position.   The same would be true for the Prog button where it would open the immediately close the memory.

So your suggested feature while no more effort than creating a frame repeater and seems like a good idea on the surface.  However, it completely boogers the ecosystem if it were put into practice.  When a frame originates from ESPSomfy RTS there is much more control but it is a new frame series and it has to be because the addresses are different.   This is why you can set the number of default repeats on the Shade/Group Config screens when ESPSomfy RTS is doing the commanding.   From the remote you hold down the button longer.

fft1010, 2024-02-09:
@rstrouse Thanks for this explanation. I think than I have to find out, how to integrate L1 into the existing group.

thanks
Juergen

rstrouse, 2024-02-09:
Open the memory by long pressing on a channel that only controls that motor.   When the motor jogs release the button.   After that switch to the remote channel that controls the group and press the prog button on that remote.   The motor should then again jog and it will be added to the group.

rstrouse, 2024-02-14:
I am going to close this for now as the feature will be available in the upcoming 2.4.0 release.   If you followed the instructions above it should have added the rogue motor to the group.

closed 2024-02-14 by rstrouse