rstrouse/ESPSomfy-RTS #262: Upgrade bricked my ESP32

Lectere, 2024-02-05:
### Hardware

ESP32

### Firmware version

2.3.1

### Application version

2.3.1

### What happened? What did you expect to happen?

Half a year ago I've installed ESPSomfy on my ESP32. 

Yesterday I've upgraded from version (I think it was something around 1.7.3) to version 2.3.1. 

Upgrade steps; se how to reproduce.

### How to reproduce it (setp by step)

```bash
1. I went with my browser to the IP of the ESP Somfy
2. I've downloaded and installed SomfyController.ino.esp32s3.bin in the Web GUI, reboot
3. I've downloaded and installed SomfyController.littlefs.bin in the Web GUI, reboot
4. After the reboot, I could not access the Web GUI
```

rstrouse, 2024-02-05:
You installed the ESP32-S3 firmware do you have an ESP32-S3.  If you were running 1.7.3 without recompiling then you most likely do not have an ESP32-S3 chip.

However, try this.  Open a browser and enter the following url.  `http://<ip address of ESP32>/downloadFirmware/ver=v2.3.3`

Lectere, 2024-02-05:
Th s3 was a copy paste error, I used the normal one...

rstrouse, 2024-02-05:
Did you run the url above from the browser?  It can take a few minutes to complete.

rstrouse, 2024-02-05:
It might help if I didn't mess up the url `http://<ip address of ESP32>/downloadFirmware?ver=v2.3.3`

Lectere, 2024-02-06:
I'm gonna try that soon, thanks for the tip, let you know...

nardev, 2024-02-09:
I didn't debug it but mine one is also pretty unstable. It's hard to open web config now.

rstrouse, 2024-02-09:
@nardev Which chip series are you running?   Also, try accessing it with the IP address to rule out any mDNS issues on the network.   If you could provide a backup that would be great to run a unit with the same chip series and data.

nardev, 2024-02-09:
This is what i get:

`$ esptool.py --port /dev/ttyACM0 flash_id
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 84:fc:e6:cf:4a:ba
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Flash type set in eFuse: quad (4 data lines)
`

nardev, 2024-02-09:
The wifi AP doesn't get created at all.

nardev, 2024-02-09:
@rstrouse as i said, i feel that it was very responsive  before where now, it barely opens the web. I did't do any investigation what's happening there. Nothing changed in regards to network.

nardev, 2024-02-09:
This lasts forever, actually, mostly fails to open after login.

[image]

rstrouse, 2024-02-09:
@nardev is up and running #236 
@Lectere you are next.  Let me know how the flashing goes.

Lectere, 2024-02-12:
I've started in the browser;

http://192.168.x.xxx/downloadFirmware/ver=v2.3.3

Nothing happens. 

I can see with the ESP tool (https://espressif.github.io/esptool-js/) that it's connected to my Wifi. 

I really need my config from it. I've downloaded it, but I cannot find it anymore. It should be on the ESP32 device...

rstrouse, 2024-02-12:
`http://<ip address>/backup` should download the file.  
The url above is also not correct it is `http://<ip address>/downloadFirmware?ver=2.3.3`

It should take a couple of minutes to finish and it needs internet access.

Lectere, 2024-02-12:
Funny, on the download URL I get this error now; 

{"status":"ERROR","desc":"Release not found in repo."}

Lectere, 2024-02-12:
On the console;

Connecting to AP
Set hostname to:ESPSomfyRTS
**********
Successfully Connected to WiFi!!!!192.168.2.107 (-58dbm)
MDNS Responder Started: serverId=DD8568
Max Heap: 110580
Free Heap: 170236
Min Heap: 167796
App Version:2.3.2
shades.cfg exists so we are using that
Reading header at 0
version:20 len:66 roomSize:29 roomRecs:0 shadeSize:276 shadeRecs:2 groupSize:194 groupRecs: 0 pos:66
Applying radio settings Setting Data Pins RX:12 TX:13
Setting SPI Pins SCK:18 MISO:19 MOSI:23 CSN:5
Radio Pins Configured!
Successfully set up the radio
Enabled receive on Pin #12 Timing: 1
Initializing RX Queue
Check github for updates...
Max Heap: 110580
Free Heap: 169648
Min Heap: 163592
Internet is Available: 2027ms
[HTTPS] GET... code: 200
[HTTPS] GET... code: 200 - -1
Max Heap: 110580
Free Heap: 163328
Min Heap: 109040
Loading file /index.html

rstrouse, 2024-02-12:
I'm sorry apparently I cannot type the version is `v2.3.3` not `2.3.3`  the correct url is 
`http://<ip address>/downloadFirmware?ver=v2.3.3`

The error message was correct and the last line in the log indicates that it cannot finish loading the index.html file from the filesystem.   This should fix the issue.

Lectere, 2024-02-12:
I've been at this project for 5 months now. Tried 5 different ESP boards. Still not going, I've bought a LilyGO T-Internet POE because wifi seems to get lost all the time...

I had it working with version 1.7.3 but wifi kept disconnecting. Maybe it was the esp32 from china...

But, the download firmware seems to go now...

rstrouse, 2024-02-12:
Let it finish it will take about 2 minutes.   Sorry this is taking so long to get going.

Lectere, 2024-02-12:
you don't have to apologize for that.  I don't know what it is. But somehow I've been busy for months. Problem is, I have kids so not a lot of time to focus. And apperently everything that can go wrong, went wrong. So anyhow,. I bought one of these LilyGO T-Internet POE. Cannot flash firmware on it... I don't know how it works. Googled it ofcourse, find a ardruno think where you can compile your own software. I don't need that, I just want the ESPSomfy image on it...

But back to the esp32 wifi, I think it's uploaded the firmware on it now. But still cannot access the webpage...

Lectere, 2024-02-12:
So, when it goes to wifi acces point mode; 

I can ping it;

[image]

But no success via the browser;

[image]

rstrouse, 2024-02-12:
I think the LilyGO T-Internet POE requires a flasher board.  It must be installed to flash the firmware and removed afterward.   

What are you seeing in the logs when the ESP32 starts up?

Lectere, 2024-02-12:
Really, a flasher board, just to get an image on it... ffs.,.. my god, another 20 euro and another week...

In the log's;

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
E (283) psram: PSRAM ID read error: 0xffffffff

Startup/Boot....
Mounting File System...
File system mounted successfully
Chip Model ESP32-
Preference IP Free Entries: 495
SECURITY   Type:0 Username:[] Password:[] Pin:[] Permissions:0
Connection Type: 1
NTP Settings 
pool.ntp.org TZ:CET-1CEST,M3.5.0,M10.5.0/3
WIFI Settings
 SSID: [XXXXXXX] PassPhrase: [XXXXXXXXXX]

Launching web server...
Creating Web MicroServices...
WiFi Mode: 0
Scanned 2 Networks...

Socket Server Started...
Max Heap: 110580
Free Heap: 178052
Min Heap: 176128
Connecting to AP
Set hostname to:ESPSomfyRTS

rstrouse, 2024-02-12:
Yeah most of the ethernet boards require such a device.

Is there actually an SSID and passphrase in there?  The next line after the Set hostname should be a series of * that indicate it is attempting to communicate with your wifi network.  And when it connects it should return an IP Address as below.

```
13:29:18.703 -> Connecting to AP
13:29:18.843 -> Set hostname to:ESPSomfyRTS
13:29:19.409 -> ****
13:29:20.904 -> Successfully Connected to WiFi!!!!192.168.1.152 (-53dbm)
13:29:20.951 -> MDNS Responder Started: serverId=49E4AC
13:29:25.943 -> Max Heap: 110580
13:29:25.943 -> Free Heap: 170004
13:29:25.943 -> Min Heap: 160296
```

Lectere, 2024-02-12:
Startup/Boot....
Mounting File System...
File system mounted successfully
Chip Model ESP32-
Preference IP Free Entries: 494
SECURITY   Type:0 Username:[] Password:[] Pin:[] Permissions:0
Connection Type: 1
NTP Settings 
pool.ntp.org TZ:CET-1CEST,M3.5.0,M10.5.0/3
WIFI Settings
 SSID: [xxxx PassPhrase: [xxxx]

Launching web server...
Creating Web MicroServices...
WiFi Mode: 0
Scanned 3 Networks...
dsf
Socket Server Started...
Max Heap: 110580
Free Heap: 177964
Min Heap: 176016
Connecting to AP
Set hostname to:ESPSomfyRTS
**********
Successfully Connected to WiFi!!!!192.168.x.x (-54dbm)
MDNS Responder Started: serverId=DD8568
Max Heap: 110580
Free Heap: 170308
Min Heap: 168084
App Version:2.3.2
shades.cfg exists so we are using that
Reading header at 0
version:1 len:0 roomSize:0 roomRecs:0 shadeSize:0 shadeRecs:0 groupSize:0 groupRecs: 0 pos:0
Invalid Shade Record Size:0
Shade config file invalid!
Applying radio settings Setting Data Pins RX:12 TX:13
Setting SPI Pins SCK:18 MISO:19 MOSI:23 CSN:5
Radio Pins Configured!
Error setting up the radio
Initializing RX Queue
Check github for updates...
Max Heap: 110580
Free Heap: 170108
Min Heap: 163972
Internet is Available: 1695ms
[HTTPS] GET... code: 200
[HTTPS] GET... code: 200 - -1
Max Heap: 110580
Free Heap: 165248
Min Heap: 112828

rstrouse, 2024-02-12:
Ok so it is not starting in AP mode.

With it connected try accessing it with the IP address and watch the log.  Interestingly it appears to not have v2.3.3 on it even after the downloadFirmware.  Which ESP32 is this?

Lectere, 2024-02-13:
I've managed to download a config file. That was around 400 bytes bug. After that, I've decided to reflash the ESP32 with a new image 2.3.3. That worked. 

I tried importing the settings from the backup, but that didn't work. So I need to re-learn the device. 

I think the hardware I had was flaky, that in combination with a half failed upgrade.. Thanks so much for the help and the great project rstrouse! I think I'm gonna try the POE version.

rstrouse, 2024-02-13:
Thanks for letting me know.   If you look in your browser downloads you may have a backup file in there that is not 0 or 400 bytes.

closed 2024-02-13 by rstrouse