rstrouse/ESPSomfy-RTS #264: Update to 2.3.3 removed all covers. Backup from within HA wont restore

danw87, 2024-02-05:
### Hardware

ESP32

### Firmware version

v2.3.3

### Application version

v2.3.3

### What happened? What did you expect to happen?

esp32-wroom-32

Prompt from HA to update RTS firmware. It completed backup then updated. Now web interface shows no covers and clicking a cover in HA has no action. 
Rolled back to v2.3.2 using downloadFirmware?ver=v2.3.2

Tried to restore the backup that was generated by the update but it states This file is not a valid backup file.

### How to reproduce it (setp by step)

```bash
1) Update from prompt in HA 
2) covers gone
3) try to restore the .backup file from ESPSomfyRTS_{ID} HA config dir
```

### Logs

```bash
None
```

rstrouse, 2024-02-05:
Post your backup file here by renaming it to .txt.  I'll be HA changes the encoding.

danw87, 2024-02-05:
[covers.txt](https://github.com/rstrouse/ESPSomfy-RTS/files/14172452/covers.txt)
Thanks, I did a quick check and couldn't see anything obvious but i dont have a good native backup to compare

rstrouse, 2024-02-05:
Sure enough the process of reading the file changed the encoding and added carriage returns at the end of every line instead of just a linefeed.  Here is your working file.

[ESPSomfyRTS.backup.txt](https://github.com/rstrouse/ESPSomfy-RTS/files/14172566/ESPSomfyRTS.backup.txt)

danw87, 2024-02-05:
Thanks, interestingly it seems its my windows machine that i copied my backup via (I must of hit save in notepad++, no shock that windows changed the new line format) that added the CR as if i open direct from HA volume its just LF and those parts match on a diff between my linux copy and the file you provided, just position and what I assume is the rolling code column have different values .

I've downloaded the file you sent, made no changes, tried to restore and get the same error.

rstrouse, 2024-02-05:
I just restored it to one of my devices.  Here is the process I did.  I clicked on the file and did not open it in any editor.  I navigated to the downloads folder in the browser then removed the `.txt` extension from it.  Then restored that file directly from the UI.
[image]

rstrouse, 2024-02-05:
In the next version of the HA plugin I managed to rip the CRLF out of the file leaving just LF.  Why Python assumes the encoding for a file that was opened in binary is a shocker.  Just write the bytes I give you </end rant>.

danw87, 2024-02-05:
It was me saving in windows that changed to CRLF.
I managed to restore the file you provided by removing .txt.
I was then able to restore my original HA backup, no modifications, by renaming from 2024-02-05T20_03_04.backup to ESPSomfyRTS.backup so seems its actually something with filename rather than content / encoding?

Either way, thank you very much for the quick responses to get it back up and running. Was the purge expected on the upgrade to 2.3.3?

rstrouse, 2024-02-05:
> Was the purge expected on the upgrade to 2.3.3?

No in fact I have 10 devices that I downgrade and upgrade all the time.  In fact I downgraded the device I restored your backup to then upgraded it again.   No goofiness there.   It does look like I have some script that looks for ESPSomfyRTS as a prefix on the file.  Sometimes I just do dumb things considering I validate the format on the ESP32.

danw87, 2024-02-05:
Odd, i just re-upgraded to 2.3.3 and all was fine so seems like it was just a hiccup during update, these things happen.

The restore operation not accepting file names from the HA backups (I call the service daily to capture rolling code changes) feels like it would be a good quality of life improvement, prevent people running into restore issues and i wouldn't have needed to bother you.

Thanks again! Awesome project

rstrouse, 2024-02-07:
Glad you got it working.  v2.4.0 of the firmware fixes the file naming for the backups and v2.4.0 of the plugin will fix the encoding that Python feels I need whether I wanted it or not.

closed 2024-02-07 by rstrouse