rstrouse/ESPSomfy-RTS #265: No MQTT between ESPsomfyRTS and HA

1cgtester, 2024-02-08:
### Hardware

ESP32

### Firmware version

V2.3.2

### Application version

V2.3.2

### What happened? What did you expect to happen?

First at all , Thx a lot to bring something like ESPsomfy on the road.  !  

But I have still a prob: 
After updating HA to  to the latest Update 2024:2 there is no Mqtt communication between HA and ESPSomfy. 
 

The ESPSomfyRTS browser is  still alive and works great.

Perhaps someone has got an idea to solve it.

best regards 

### How to reproduce it (setp by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-02-08:
First let me say that my first choice for communication with HA would not be MQTT.  There is a direct socket connection with the plugin and it manages the devices so much better.

If you watch the serial console you should be able to see whether it successfully connected to the MQTT broker.   In looking in their blog there are a few others that are experiencing weirdness with MQTT and 2024:2.

Also check the connection data is correct in ESPSomfy RTS in the network section on the MQTT tab.   If you are running MQTT explorer you can also check to ensure the MQTT broker is up and ESPSomfy RTS is publishing to it.

1cgtester, 2024-02-08:
thanks for the quick reply,
I must have accidentally changed a number in MQtt.

That was my mistake, now it works.
Thanks

rstrouse, 2024-02-08:
Glad you got it working!

closed 2024-02-08 by rstrouse