rstrouse/ESPSomfy-RTS #266: Strange issue while adding new remotes.

nardev, 2024-02-09:
### Hardware

ESP32

### Firmware version

V2.4.0

### Application version

V2.4.0

### What happened? What did you expect to happen?

Despite all the effort to "migrate" remote settings with the backup, it didn't work out.

After flashing SomfyController.onboard.esp32.bin to ESP32 Wroom, i got the Web Interface, i tried restoring previous backup and after reboot it only created the old remotes.

The wiring of the board is 1:1 same as it was while it was working.

This is what fails:

- I tried SCAN FREQ and i saved what appeared as a remote feedback
- Old remotes that were created from the backup file do not work
- I tried pairing a remote again by creating new remote and by doing as instructed and as before:
    - Press PROG on my remote until it jogs, press PROG on the web interface and when it jogs, press "paired". However, it did jog ok for both times but after that, the new remote simply doesn't work.

So, it was evidently paired but for some reason, it can't control it.

In the radio log i ca catch all commands from the remote.

Transceiver settings as below:

Base Freq 433.420
RX Bndw 97.96
Freq Dev 11.43
TX Pow 10dBm
-68 dBm

The scan freq does catch frequency, keeping it the same or increasing it a bit, doesn't change anything.

Any clue?

### How to reproduce it (setp by step)

```bash
Just tried adding new remotes. All wiring same as before while it was working.
```

rstrouse, 2024-02-09:
Check the rolling code value.   Does it match what is in the backup?

nardev, 2024-02-09:
Yes it does, timing and rolling code is the same,.

Now i managed to "unpair" two of those and i'm trying to add them as new remotes, but no luck.

rstrouse, 2024-02-09:
Try a new address.  This sounds like the current address stored in the motor does not match the current rolling code.  If the codes don’t match it will not paire or unpair.

nardev, 2024-02-09:
I did move +50 in addresses, nothing changes. I'll try one more time with another address.

P.s. yes, once i removed the remotes, new address again took the same number. Moving new remotes makes sense.

nardev, 2024-02-09:
:( No luck. Just added new, very high number, pairing went well but commands do not work.

when i say "went well" it means, that shade jogged on both remote and web remote

nardev, 2024-02-09:
Shit, somehow i managed to put two in pair now... Too much playing with this :(

rstrouse, 2024-02-09:
If you have been pairing up a bunch of remotes you may need to factory reset the motor.   I had one that obviously had several remotes paired before it was installed.

nardev, 2024-02-09:
@rstrouse thank you.

I'll probably do that tomorrow.

nardev, 2024-02-09:
Eventually, the ESP32 board i was using was actually bricked. It was not possible to erase flash or write new FW

rstrouse, 2024-02-09:
What is the error message you are getting?  If it isn't a chip failure are you sure you are flashing the right firmware?  Is the serial port coming alive.

nardev, 2024-02-09:
esptool doesn't report any error but flash erase finishes in 0.0, also flashing is very quick. But even arduino IDE upload failed.. it's already in trash. Who cares.

rstrouse, 2024-02-10:
That definitely sounds like either a hardware failure or the SPI interface for the comm serial port went bad.

rstrouse, 2024-02-14:
@nardev I am closing this for now.   Let me know if you need further assistance.

closed 2024-02-14 by rstrouse