rstrouse/ESPSomfy-RTS #267: [INFO] ESP32 firmware

chemelli74, 2024-02-10:
### Describe you new feature you'd like

I have the following board: [ESP32 NODEMCU DevKit C (with CP2102) from Az-Delivery](https://www.az-delivery.de/en/products/esp32-developmentboard)

Not sure which is the best firmware to select.
Can you please give an advise ?

rstrouse, 2024-02-10:
Select the firmware from `SomfyController.onboard.esp32.bin.zip`.  This is a WROOM/WROVER chip which is part of the ESP32 series.  

The confusing part is that there is there a bunch of other chip series that all start with ESP32.  For instance ESP32-S2, an ESP32-S3, EPS32-C2 ...etc are all ESP32s but are a completely different chip series.  Given the way Espressif names things I would not recommend letting them name your children.

rstrouse, 2024-02-10:
I added a chart to the onboarding instructions.

closed 2024-02-10 by rstrouse

chemelli74, 2024-02-11:
Thank you so much for the quick feedback; the new chart will help for sure others as well.

I have one more question: would be possible to consider making your work a plugin for [esphome](https://github.com/esphome/esphome) ?
I ask because I would like to have a single device able to control both Somfy and a IR climate located on the same terrace.

Hopefully I didn't abuse of your time and kindness

rstrouse, 2024-02-12:
I have no desire to create this as a plugin for ESPHome.  When I started this I took a long hard look at what it was all about and decided that it would scatter my focus on what I wanted it to do.  While ESPHome has a strong focus on sensors and other fun stuff, by now I would be rewriting it to unload all of the ESPHome from it.

chemelli74, 2024-02-12:
I perfectly understand your point of view.

Is just a pity not to be able to share the hardware for more uses.
Not for the costs, but for the logistic: taking more power to the terrace, need to position 2 boxes that takes more space and are less esthetical.

Thank you anyway for taking the time to reply.
Much appreciated.