rstrouse/ESPSomfy-RTS #268: Static IP does not work on ETH

Jezza34000, 2024-02-10:
### Hardware

ESP32 WT32-ETH01

### Firmware version

v2.3.3

### Application version

v2.3.3

### What happened? What did you expect to happen?

Hi Robert,
I notice the same problem as this issue : https://github.com/rstrouse/ESPSomfy-RTS/issues/193
The fixed IP address is only respected in Wifi, as soon as I switch to ETH it is assigned from the DHCP
Static reservation by mac address is disabled on my Freebox (router)

### How to reproduce it (setp by step)

```bash
1. Disable "Acquire IP adress autmatically (DHCP)"
2. Set static IP
3. Connect through LAN (=> IP is from DHCP)
4. Switch to Wifi  (=> IP is from ESPSomfy RTS's settings)
```

rstrouse, 2024-02-10:
I'll have a look at the ethernet library to see if there are any quirks in it.  The interweb is riddled with puzzles regarding the netif for ethernet on esp32.  Like all things interweb much of it is nonsensical.

rstrouse, 2024-02-10:
Install the v2.4.0 pre-release.  

I looked at the code in the ethernet libraries and it appears the initialization of the board resets everything to DHCP.   So this means that it starts the DHCP module whether you want it or not.   So not to be outdone, I set it back after the board initialization before it has a chance to start it up.  I can see why all the confusion on the interweb.  This is a departure from the Wifi adapter.

rstrouse, 2024-02-12:
I'll have another look this evening.

Jezza34000, 2024-02-13:
Tested an approuved. 2.4.0
Wifi => OK
ETH => OK
Many reboot => OK

closed 2024-02-13 by Jezza34000