rstrouse/ESPSomfy-RTS #269: Checkbox "Fallback to Wireless" does not save it state

Jezza34000, 2024-02-10:
### Hardware

ESP32 WT32-ETH01

### Firmware version

v2.3.3

### Application version

v2.3.3

### What happened? What did you expect to happen?

Hi Robert,
Checkbox "Fallback to Wireless" does not save it state.
I can use either ethernet or wifi, both configuration are working well

### How to reproduce it (setp by step)

```bash
1. Configure ETH/WIFI
2. Check "Fallback to Wireless"
3. Click on "Save Adapter"
4. Reboot > and then "Fallback to wireless" is unchecked
```

rstrouse, 2024-02-10:
I'll have a look at the ethernet library for quirks.  The software is setting it but the interweb is riddled with inconsistencies regarding it sticking on the PHY.

rstrouse, 2024-02-10:
Install the v2.4.0 pre-release.  The function now properly persists the fallback condition.

Jezza34000, 2024-02-12:
Working OK, fixed 2.4.0