rstrouse/ESPSomfy-RTS #272: Website of ESPSomfy-RTS slow and not fully loading

fft1010, 2024-02-12:
### Hardware

ESP32

### Firmware version

V2.4.0

### Application version

v2.4.0

### What happened? What did you expect to happen?

I just read the logs this morning, and saw what I pasted below. I than wanted to check the original ESPSomfy-rts webpage, and all I saw was this:

[image]

After reboot (plug off/on) it was fast again. For testing I deleted the connected repeater remote, and will let you know if this changes anything. 

### How to reproduce it (setp by step)

```bash
Nothing special.
```

### Logs

```bash
2024-02-11 20:51:40.652 ERROR (MainThread) [homeassistant.components.update] Error while setting up espsomfy_rts platform for update
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 344, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/espsomfy_rts/update.py", line 28, in async_setup_entry
    async_add_entities([ESPSomfyRTSUpdateEntity(controller)])
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/espsomfy_rts/update.py", line 42, in __init__
    self._attr_unique_id = f"update_{controller.unique_id}"
                                     ^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/espsomfy_rts/controller.py", line 217, in unique_id
    return f"espsomfy_{self.server_id}"
                       ^^^^^^^^^^^^^^
  File "/config/custom_components/espsomfy_rts/controller.py", line 212, in server_id
    return self.api.server_id
           ^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/espsomfy_rts/controller.py", line 435, in server_id
    return self._config["serverId"]
           ~~~~~~~~~~~~^^^^^^^^^^^^
KeyError: 'serverId'
2024-02-11 20:51:40.655 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up espsomfy_rts platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 344, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/espsomfy_rts/sensor.py", line 96, in async_setup_entry
    new_entities.append(ESPSomfyDiagSensor(controller=controller, cfg=ESPSomfyDiagSensorDescription(
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/espsomfy_rts/sensor.py", line 115, in __init__
    self._attr_unique_id = f"{cfg.key}_{controller.unique_id}"
                                        ^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/espsomfy_rts/controller.py", line 217, in unique_id
    return f"espsomfy_{self.server_id}"
                       ^^^^^^^^^^^^^^
  File "/config/custom_components/espsomfy_rts/controller.py", line 212, in server_id
    return self.api.server_id
           ^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/espsomfy_rts/controller.py", line 435, in server_id
    return self._config["serverId"]
           ~~~~~~~~~~~~^^^^^^^^^^^^
KeyError: 'serverId'
```

rstrouse, 2024-02-12:
This tells me that the device was stopped at that moment and HA was started without the ESPSomfy RTS being up.   There is a mod coming to the HA component that waits until it can secure a connection before allowing the entities to be initialized.

fft1010, 2024-02-15:
Hi @rstrouse ,
does this mod affect these cards as well?
[image]

They are IMHO a little too "smart". As sometimes (very rare) not all shades listen, and this card assumes, the share is up, but it isn't. So pressing down/stop/up solves the problem, but a checkbox in configuration like "behave as remote" (and than always all up/stop/down function would be great IMHO.

If this can be done, than I could nod find it how. But I have limited HA knowledge, a hint what direction to google would be great than.

thanks
Juergen

rstrouse, 2024-02-16:
If I could I would not allow the buttons to be disabled but that is a ride-or-die for the home assistant folks.   I don't think these cards are smart at all.  If they were I would have long ago did exactly what you are referring to.  And yes the plugin does wait to load the entities behind these cards until the ESPSomfy RTS device can be contacted.  This allows the boot to occur in any order.

rstrouse, 2024-03-10:
@fft1010 has your original issue been solved with the things we have been through?  I can't do anything with the cards as that is core ui in HA.

fft1010, 2024-03-11:
Hi @rstrouse,
no, it was not solved. Once in a while this shade does not listen to the signal. But I might have another idea, which might work, but definitely is a "work around". In an another installation of shades with a Bosch System, it looked to me, as if the Bosh "tell", when the shade is up. If this is right, that I assume ESPSomfy RTS also "tells" HA, that the shades are open/closed as well. So why not make an option in ESPSomfy, that enables the system to "lie" a little bit. So instead of telling Shade "open", ESPSomfy only tells "99%" open, and instead of closed it says "99%" closed. (or 1% open), whatever HA needs to make the Up or down "work" always.
I know this is somehow clumsy, but IMHO the shades might look closed in the icons as well, and we are closer to a remote, and HA does not have to change anything. Sideeffects might be, that if users try to catch the "closed" or "open" in HA, they won't get it. And switches will not show the correct status. But _maybe_ this is the better choice in such environments that I am living in.
Just an idea.
many thanks
Juergen

rstrouse, 2024-03-23:
I think you will get much more reliable comms if you add a repeat or two to your shade definitions.  It is common especially if you have battery operated motors for them to be a little bit sleepy.  I experimented with the 1% and 99% thing and I did not like the results.  In fact it created issues with other things such as the sensors.

fft1010, 2024-03-23:
Hi @rstrouse ,
Thank you for trying it out. I was not sure about the side effects, and I donno, if it would make sense with a clear warning. Anyway, a „simply“ remote would be best, but if this won‘t work, I live with what I have, which is much better, than what I had before.

many thanks
Juergen

closed 2024-03-23 by fft1010