rstrouse/ESPSomfy-RTS #276: Help debugging blind that does not stop at the requested position

nbarrientos, 2024-02-14:
### Hardware

LilyGO T-Internet POE

### Firmware version

v.2.3.3

### Application version

v.2.3.3

### What happened? What did you expect to happen?

Hi,

This is not exactly a bug report, rather a call for help debugging.

I've got a HA automation that opens a blind to a certain position (95% open, following HA's terminology) when a given condition is met. What I see _sometimes_ is that the blind is fully opened instead, however it reports as 95% open in HA (and 5% closed in ESPSomfyRTS). My guess is that either the `My` pulse to stop the blind at the requested position is either not sent (because HA does not request it, or because it does but effectively never leaves ESPSomfyRTS), lost in the air, or just ignored by the motor.

May I ask for help on how can I debug what's going on?

HA integration version: v.2.3.3

Thanks.

### How to reproduce it (step by step)

```bash
* Set blind position to a specific position.
```

rstrouse, 2024-02-14:
Your up and down timings are likely not correct for the shade definition.  Enter these in milliseconds 1/100th of a second.

nbarrientos, 2024-02-14:
Thanks.

That would indeed explain it if it happened consistently, would it? The issue only happens sometimes, though. 

I believe the timings are correct (address removed on purpose):

[image]

I measured the `Down Time` by fully opening the blind, then pressing down and waiting until the motor stopped by itself (when the blind is fully closed). Vice-versa for the 'Up Time`.

Another data point is that I'm unable to reproduce the problem by setting the position using ESPSomfyRTS (using the Target Position slider), although I managed to reproduce it manually using HA.

It does not make much sense though, reading the code of the HA integration it seems that HA asks ESPSomfyRTS "set blind to position X", so the actual `My` press to stop the blind is ultimately set by ESPSomfyRTS, no matter if I interact with the blind using HA or ESPSomfyRTS directly. It has to be possible to reproduce the problem using ESPSomfyRTS directly. I'll insist and come back.

rstrouse, 2024-02-14:
Add 1 repeat. Just to make sure the my command is making it to the motor.

nbarrientos, 2024-02-16:
I'll try that too and keep an eye on it. Please keep this issue opened for a few days.

rstrouse, 2024-02-24:
@nbarrientos did the repeat fix your issue.

nbarrientos, 2024-02-25:
I haven't had time to do more through checks, so I'm relying only in the morning automation that opens the blinds. So far the offending blind has been in the desired position every morning so in principle it has helped. Feel free to close, I can reopen if I spot any other discrepancy. Thanks!

rstrouse, 2024-02-25:
Thanks for closing the loop.   I suspected a repeat or two would solve the issue.

closed 2024-02-25 by rstrouse