rstrouse/ESPSomfy-RTS #277: New blinds Wirefree solar RTS

luckyzor, 2024-02-14:
### Hardware

ESP32

### Firmware version

v2.3.3

### Application version

v2.3.3

### What happened? What did you expect to happen?

Hello,

I've installed new solar blinds, they are at the exterior of the house (they are solar)
I think I may have issues with the signal, but whats is strange: after clicking for multiple times the command will work and all the other commands sent right away work too (most of the times). If i wait 5 sec, it stops working again, I need to send the command multiple times again. this is really random..

the esp + E07-M1101D-SMA + Eightwood Antenna 433mhz 3dbi
Installed near the garage motor
[image]

logs at 8 m from the antenna with 2 walls (the motors are at the exterior of the house, so its worse)

```
[{"encKey": 169,
  "address": 8098888,
  "rcode": 121,
  "command": "Up",
  "rssi": -74,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [2568,2472,2521,2458,4825,1270,1308,1245,1324,1225,1318,609,686,1246,671,613,1312,593,695,587,686,1242,1321,1240,680,593,695,585,1336,593,669,610,685,1251,1293,1268,676,595,673,613,664,623,671,603,678,607,1308,598,675,1241,1329,1258,1310,593,670,613,687,1225,1319,609,683,1242,1323,614,666,603,680,1241,673,599,700,583,1334,1221,1334,1244,1324,1241,1319,1241,1308,1251],
  "time": "2024-02-14T14:23:19.603+0100"
},
{"encKey": 169,
  "address": 8098888,
  "rcode": 121,
  "command": "Up",
  "rssi": -74,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [685,27441,2563,2441,2542,2441,2563,2455,2538,2448,2564,2446,2537,2445,2560,2443,4863,1212,1339,1231,1339,1222,1329,598,696,1227,686,600,1323,610,659,603,677,1245,1334,1244,671,609,664,621,1316,589,686,599,671,1251,1312,1244,700,591,688,591,691,584,684,604,679,593,1321,615,681,1243,1305,1248,1320,604,688,596,674,1240,1330,604,679,1242,1325,584,683,615,682,1242,677,589,685,593,1319,1239,1330,1256,1311,1245,1305,1247,1320,1242],
  "time": "2024-02-14T14:23:19.751+0100"
},
{"encKey": 170,
  "address": 8098888,
  "rcode": 122,
  "command": "Up",
  "rssi": -74,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [2570,2441,2547,2444,4857,1222,1320,1243,1321,1238,1334,1246,1310,1244,1324,598,689,601,678,1242,1301,607,681,616,658,1245,1326,599,671,599,703,1216,1327,606,681,590,684,1243,674,616,687,596,669,606,1306,622,670,1230,1337,1242,1304,596,689,606,681,1234,1322,612,685,1217,1322,612,684,590,672,1252,686,601,674,597,1325,1249,1318,1241,1301,1259,1313,1244,1306,1246],
  "time": "2024-02-14T14:23:21.747+0100"
}]
```

I really don't understand How can improve the signal to the motors.
can anyone help me?

thank you in advance

### How to reproduce it (step by step)

```bash
-
```

luckyzor, 2024-02-14:
more info: with the smoove origin RTS button near the antenna, 20cm +-, I got 45dBm and when I pushed the button the shutter worked. so the button can handle this distance.
can this be an issue with the esp + E07-M1101D-SMA ?

rstrouse, 2024-02-14:
Most likely it is the antenna.   I believe that is the exact mighty mule antenna that I bought a long time ago and had horrible results.  I think has some weird tuning.

rstrouse, 2024-02-14:
There are a couple of things that could be in play here.   Antennas can often have hot and cold spots based on their design.  They never radiate evenly.   I noticed the garage door (big metal ground plane) this too can change the dynamics of the antenna.  

However, I would like you to double check something for me.   Make sure any addresses you see in the logs do not match any addresses that you have set for the motors.   The symptoms you are describing sound like ESPSomfy RTS becomes successful when its rolling codes catch up with the remote.

luckyzor, 2024-02-14:
I bought this antenna after I had the same results with the default antenna (the little one)
the most weird, I tested with espsomfy device near the shutter and had the same results... 
concerning the addresses: 

[image]

[image]

rstrouse, 2024-02-14:
How many repeats do you have set up for the motors.   Are they sleeping after 5+ seconds and slow to get out of bed?

luckyzor, 2024-02-14:
No repeats, because with the smoove button, I only need to click once.
I don't know if there are any sleep time, I did not configure the shutters, but as the shutters only need one click from the smoove button, don't no if there are any sleep timer.

rstrouse, 2024-02-14:
1 click often results in many repeats.  This happens on very rapidly for instance the pulses above show 5 repeats.  Set it to 2 and see if this makes a difference.  It doesn't mean that it is commanding twice it means that after the wakeup pulses it will send the same command multiple times.  This is commonly needed with battery motors.   They tend to go to sleep to conserve energy.  Since we are talking about very rapid pulses they can drop some of them in their slumber but often pick them up on the next frame.

luckyzor, 2024-02-15:
thank you for your help,
I tried with 2 times and it worked for 4 of 5 shutters, set it to 3 repeats can to anything bad?

now I have another issue, when someone open the shutters with the smoove button, the espsomfy don't update the state of the shutter, are there any way to achive this?

thank you again for your help

closed 2024-02-17 by luckyzor