rstrouse/ESPSomfy-RTS #278: Warning triangle on the blind icon

svgenov, 2024-02-16:
### Hardware

ESP32

### Firmware version

v.2.3.3

### Application version

v.2.3.3

### What happened? What did you expect to happen?

Congratulations for the wonderful work. I'm using ESP32 Wroom 38Pin plus CC1101 433MHz module. I was able to setup everything from the first try (I have 6 RTS blinds), however, one of the blinds is displayed with a warning triangle. I found no information about this problem in the docs (or maybe I did not search hard enough). It was paired like all the rest, following the documentation, the remote is also paired, and it works perfectly. The only issue is actually the icon with the warning triangle, which bugs me. I was not able to find any logs of the software running, so I attach the screenshot, my backup plus the logs from the radio (when I work with the blind in mention with the remote). I tried to unpair/pair it numerous time, deleting it and adding it again. Actually there was such problem with another blind, which disappeared after the first unpair/pair loop.

[image]
[image]

### How to reproduce it (step by step)

```bash
I just go to home and see the blind with the exclamation mark.
```

### Logs

```bash
[{"encKey": 171,
  "address": 3865803,
  "rcode": 596,
  "command": "Down",
  "rssi": -42,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [101541,610,67719,10072,97358,2564,2558,2538,2570,4870,1280,1312,1268,1287,1269,1310,1272,675,611,660,621,666,632,661,641,1301,627,659,615,679,618,674,613,660,1272,674,613,660,623,1319,613,684,621,668,1260,1304,1274,1315,1268,659,614,1315,1273,662,624,1291,1289,1301,628,660,616,677,1271,659,615,1316,622,658,634,665,1270,660,632,640,644,1307,625,668,627,659,614,664,1275,1310,1269,660,630,664,620,1306,1265,674,613],
  "time": "2024-02-16T19:44:24.760+0200"
},
{"encKey": 171,
  "address": 3865803,
  "rcode": 596,
  "command": "Down",
  "rssi": -41,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [635,32368,2551,2573,2526,2565,2552,2551,2554,2562,2524,2576,2529,2585,2522,2575,4860,1269,1307,1272,1324,1242,1318,1268,671,620,654,615,665,26,622,668,631,1312,618,670,636,662,624,665,627,659,1257,666,629,661,616,1328,625,659,615,679,1243,1335,1245,1327,1250,669,637,1312,1244,684,609,1304,1275,1311,618,676,614,685,1247,675,612,1310,618,677,636,659,1271,653,614,685,621,1294,635,664,618,674,614,661,1272,1326,1254,668,609,662,642,1299,1257,684,619],
  "time": "2024-02-16T19:44:24.865+0200"
},
{"encKey": 172,
  "address": 3865803,
  "rcode": 597,
  "command": "Up",
  "rssi": -45,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [30400,10076,61071,36292,2573,2544,2544,2570,4870,1278,1283,1298,1285,1270,656,641,1290,646,651,1266,1322,609,661,640,648,645,654,1265,671,622,656,639,640,647,1295,638,665,620,668,634,638,1290,1293,1265,671,621,656,640,1293,1270,1314,619,671,635,661,615,671,1272,662,625,644,650,1283,1272,1326,1251,1320,614,666,644,646,634,661,1275,1297,1263,670,634,1286,1293,658,615,1305,1274,659,640,641,631,1318,1264],
  "time": "2024-02-16T19:44:27.243+0200"
}]
```

rstrouse, 2024-02-16:
Go into the virtual remote and send a sensor command without the wind or sun selection clicked.  This indicates a windy condition which you likely sent from the virtual remote earlier.  If this were an awning motor it will retract until it is overridden.

svgenov, 2024-02-16:
Thank you so much. Yes, possible. I don't know however, why this didn't go away when I unpaired, deleted the shade and paired it again. Anyways, sorry for the fuss, its definitely not a bug. Thank you for the prompt reply, and for the amazing job you did with the software and the hardware instructions. It's awesome.

rstrouse, 2024-02-16:
I am going to close this for now.  Let me know if you need anything else.

closed 2024-02-16 by rstrouse